gazebosim/gz-sim #364: ign-gazebo3 DiffDriveTest.SkidPublishCmd test failing

jacobperron, 2020-09-22:
https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-bionic-amd64/55/testReport/

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-bionic-amd64/ign-gazebo/test/integration/diff_drive_system.cc:306
Expected: (poses[0].Pos().Y()) < (poses[3999].Pos().Y()), actual: 2 vs 2
```

jacobperron, 2020-09-22:
Same failure on macOS: https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/59/testReport/

chapulina, 2020-09-22:
This should be fixed by https://github.com/ignitionrobotics/ign-gazebo/pull/362, which forward-ports https://github.com/ignitionrobotics/ign-gazebo/pull/357

chapulina, 2020-10-10:
The build is blue-green: https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-bionic-amd64/

closed 2020-10-10 by chapulina