gazebosim/gz-sim #367: Allow deselecting entities in entity tree with CTRL

sloretz, 2020-09-23:
`CTRL` clicking can be used to select entities, but not deselect them. Other programs do allow deselecting by `CTRL` left clicking again (example [Microsoft Excel](https://support.microsoft.com/en-us/office/deselect-a-selection-d1b56f30-7d63-4a64-8f0b-da4958204b52))

[image]

herobank110, 2020-09-25:
Hello, could I take a look at this issue?

chapulina, 2020-09-25:
@herobank110 , feel free to pick it up, I don't think anyone is working on this. Thanks!

mjcarroll, 2021-05-27:
Thanks @WagnerMarcos ! I have assigned to you and moved to "In Progress".  Feel free to tag me as a reviewer when your PR is up.

WagnerMarcos, 2021-06-02:
In this Issue the idea is that when holding ctrl and using left click on a selected entity from the entity tree, it gets deselected. In [EntityTree.qml](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/plugins/entity_tree/EntityTree.qml#L217) this click gets detected and then calls [EntityTree.cc](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/plugins/entity_tree/EntityTree.cc#L375) to send a selection event. The visual part in qml is simple and straightforward, but the C++ is taking more time than expected.

I'm trying to get right the extent of the Issue, as the events gets filtered in [Scene3D](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/plugins/scene3d/Scene3D.cc#L2836), [Component Inspector](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/plugins/scene3d/Scene3D.cc#L2836) (which could be ignored at the moment), [Joint Position Controller](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/plugins/joint_position_controller/JointPositionController.cc#L306) (also ignorable) and the [AlignTool](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/plugins/align_tool/AlignTool.cc#L484). 

For this Issue I'm thinking I have some alternatives I could follow. The first one would be to define a new [event](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/include/ignition/gazebo/gui/GuiEvents.hh#L90) and make parallelism between the selection/deselection. A similar one to this would be to have some boolean to tell if the clicking interaction with the entity was for selecting or deselecting.

Another alternative would be to ask in the plugin that filter this event when a selection was made, if that item was already selected, and if so, to deselect it. But for this there could be it should be checked if the selection was made with the ctrl key pressed, and that might be messy.

Things to consider would be how to do the same when an entity gets click on the gazebo scene, to also update the model correctly.

For the Component Inspector, would it be ok to set it kNullEntity when deselecting? (This way, it would show the world properties)
And for the AlignTool it only updates [if the process is not user created](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/plugins/align_tool/AlignTool.cc#L507), so no changes might be needed there.

francocipollone, 2021-06-03:
I assume that the behavior of this feature, which was requested for the entity tree, should be paired with `Scene3D` plugin of course. So as there is a consistency between which entity is selected in the entity tree and the highlighted entities in the scene.

From here, some expected behaviors are detached. 
 - 1. First of all, we have the visual upgrade and the actual request of the ticket: Deselecting entities in the entity tree by ctrl-clicking its entities. 
 - 2. Propagation to `Scene3D`:
      - When deselecting an entity in the entity tree should also "lowlight" the entity in the Scene that previously was highlighted.
 - 3. Propagation to `EntityTree` from other plugins.
      - `Scene3D` should allow you to deselect the entities in the Scene by ctrl-clicking on them.
      - `EntityTree` should be reflected with the current changes in the Scene (selection).

I think that focusing on how Scene3D and EntityTree deal with this new feature is the key to this issue. 
From what I saw in `ComponentInspector`, `JointPositionController` and `AlignTool` I don't think that this new deselection feature in the `EntitiyTree` should necessary enforce to change something in the behavior of those plugins, at least for the moment.

scpeters, 2021-06-03:
yes, it turns out this is a bit more complicated than I had anticipated; I'm going to remove the "good first issue" label

scpeters, 2021-06-03:
> 1. First of all, we have the visual upgrade and the actual request of the ticket: Deselecting entities in the entity tree by ctrl-clicking its entities.
> * 1. Propagation to `Scene3D`:
>      
>      * When deselecting an entity in the entity tree should also "lowlight" the entity in the Scene that previously was highlighted.
> * 1. Propagation to `EntityTree` from other plugins.
>      
>      * `Scene3D` should allow you to deselect the entities in the Scene by ctrl-clicking on them.
>      * `EntityTree` should be reflected with the current changes in the Scene (selection).

I think this proposal makes sense, but there is currently a large effort to move much of the functionality from these plugins to ign-gui (https://github.com/ignitionrobotics/ign-gui/issues/137), so I think it may be best to hold off on this work for now. It can be revisited after the larger migration is done

WagnerMarcos, 2021-06-04:
I'll put it on hold then and work on another issue then. I'll try to keep up with that migration so I know how it affects the issue for when I come back to it.