gazebosim/gz-sim #368: Add visual clue that Plugins menu is scrollable

sloretz, 2020-09-23:
(Ignition Citadel 3.3.0) It looks like the Plugins menu is missing the `Transform Control` plugin, but what is actually happening is the vertical size of the Plugins menu is limited and the plugin menu is scrollable.

It would be helpful if there was a visual cue that the menu was scrollable. Maybe an always shown scrollbar, or a gradient at the bottom if there are more choices?

[image]

Here's what each might look like:

[image]
[image]

chapulina, 2020-09-23:
Yeah I've seen people caught by this before, thinking some plugin wasn't available when all they needed to do was scroll.

I think we can just make the scroll bar always visible. The fix would go on `ign-gui`, I think all that's needed is `policy: ScrollBar.AlwaysOn` [here](https://github.com/ignitionrobotics/ign-gui/blob/22434e23832760130c88d38d7c569a049dcec65c/include/ignition/gui/qml/PluginMenu.qml#L47-L49).

chapulina, 2020-10-02:
Done on https://github.com/ignitionrobotics/ign-gui/pull/134

closed 2020-10-02 by chapulina