gazebosim/gz-sim #369: Hitting `Esc` does not return to Selection mode

sloretz, 2020-09-23:
The Manipulating Models tutorial says `Esc` returns to select mode from any other mode, but this does not appear to be the case.

> You can always return to selection mode from any other mode by pressing `Esc`.

https://github.com/ignitionrobotics/docs/blame/master/citadel/Manipulating_models.md#L42 
https://ignitionrobotics.org/docs/citadel/manipulating_models#select-mode

Hitting `Esc` in `Translate mode` or `Rotate mode` deselects entities that are being acted upon, but it never returns to `Select mode`. `Esc` as deselect feels pretty intuitive to me - maybe the line about returning to `Select mode` should be removed from the tutorial?

[image]

chapulina, 2020-09-23:
> but it never returns to Select mode

It does for me if the 3D scene has focus, but not if the entity tree has focus. I think we could just make sure that works. And also mention on the documentation that esc also deselects all entities? CC @JShep1 

[image]

sloretz, 2020-09-23:
> It does for me if the 3D scene has focus, but not if the entity tree has focus.

Interesting! If I add shapes to `empty.sdf` then it works for me with the 3D scene in focus, but if I launch `shapes.sdf` and add the `Transform Control` plugin then I get the behavior in the issue.

JShep1, 2020-09-23:
> Interesting! If I add shapes to `empty.sdf` then it works for me with the 3D scene in focus, but if I launch `shapes.sdf` and add the `Transform Control` plugin then I get the behavior in the issue.

Yeah, so just for some background, this issue is likely boiling down to how QML captures input.  Whichever section is currently focused will accept the input and the other sections will not observe this input, I suppose this is a QML way of optimizing its front-end, but it has proven to be sort of a nuisance in cases like these.  Currently, the method of the focus transitioning to the 3d scene is simply hovering your mouse over that area, so no action will occur on a button press unless the mouse has at least hovered over the 3d scene first (possibly we should add some documentation about this, I tried to fix this awhile ago and honestly couldn't find the best solution which focuses the 3d scene at all times, it may be worth re-visiting).

I see what you mean with the transform control not returning to select mode when the escape button is pressed when you load in the Transform Control via the plugin menu, I'll need to look into this some more.

JShep1, 2020-11-24:
This issue should be solved in https://github.com/ignitionrobotics/ign-gazebo/pull/466 .  The reason this was occurring is because the `TreeView` in the `Entity Tree` is capturing the escape key press under the hood.  The transform control plugin was using a qml shortcut for escape which under the hood is waiting for an escape key press.  Since the `TreeView` was capturing the escape key press event, the Transform control wasn't seeing the key event and wasn't transitioning the plugin back to the select state.  The reason entities are deselected is because `Scene3D` is waiting for an escape key **release** to deselect entities.  In the above mentioned PR, I moved transform control away from qml `Shortcut`s and also set the plugin to transition to the select state if an escape key **release** event is detected.

closed 2020-12-28 by JShep1