gazebosim/gz-sim #370: View angle plugin buttons are close to the background color, making them hard to differentiate

sloretz, 2020-09-23:
(Ignition Citadel 3.3.0) I'm having trouble telling which view angle button is which (without hovering over it and reading the tool tip text). I can see the orange face just fine, but the gray edges are very difficult for me to see (could it be my monitor is not calibrated correctly?).

[image]

A color picker shows the background is #fafafa (left) while the edge color is #f3f3f3 (right)
[image]

chapulina, 2020-09-23:
Just use dark mode :innocent:  I'm kidding, we've been trying to make sure all icons show up on both light and dark modes, but I think we just reused those ones from Gazebo-classic for expediency.

I think we could use a darker grey on the lines, and possibly make them thicker? The icons are [here](https://github.com/ignitionrobotics/ign-gazebo/tree/master/src/gui/plugins/view_angle). 

Any thoughts, @ColeOSRF?

herobank110, 2020-09-25:
Hi, Can I make those adjustments to those icons?

chapulina, 2020-09-25:
@herobank110 , you're welcome to make the changes and open a pull request, thanks!

herobank110, 2020-10-19:
I have made darkened variants of the icons for preview below. Any darker than 45 had too little contrast with the dark theme background. Let me know what you think is best.

[image]

chapulina, 2020-10-19:
Thanks, @herobank110 ! 0 is hard to see on the light background on my monitor. I'd go with 25.

herobank110, 2020-10-21:
please see #426

closed 2020-10-27 by chapulina