gazebosim/gz-sim #371: About dialog with versions and links to online resources

chapulina, 2020-09-24:
Most applications will have an "About" window that shows the exact version that the user is running. Here's Gazebo-classic's one for example:

[image]

This is handy for users that are not on a terminal or aren't running with enough verbosity to see the version printed.

I also think it would be handy to print the version of all Ignition libraries that Gazebo is using.

And we should have links to online resources too.

Ignition GUI has a [silly About dialog](https://github.com/ignitionrobotics/ign-gui/blob/22434e23832760130c88d38d7c569a049dcec65c/include/ignition/gui/qml/Main.qml#L221-L245) that could be updated as well:

[image]

chapulina, 2020-10-21:
Some tips on getting started with this issue:

1. I recommend targeting Ignition Citadel, which is branch `ign-gazebo3`
1. The dialog code can go directly into [GazeboDrawer.qml](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/resources/GazeboDrawer.qml)
1. We can create a new C++ class similar to [GuiFileHandler](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/src/gui/GuiFileHandler.hh) that will provide version information to the QML code. To start with, it could just provide [IGNITION_GAZEBO_VERSION_FULL](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo3/include/ignition/gazebo/config.hh.in).

atharva-18, 2021-01-30:
@chapulina 
I have created a dialog box inside `GazeboDrawer.qml`

[image]

Currently, I'm providing the version number and header using `IGNITION_GAZEBO_VERSION_HEADER`, and the dialog content is:
```
std::string(IGNITION_GAZEBO_VERSION_HEADER)
<div>
  <table>
    ...
  </table>
</div>
```
However, the hyperlink to the tutorials is specific to Citadel, which might cause an issue in a future version, when the version name changes. Is there a general way to provide the hyperlinks?

chapulina, 2021-02-01:
This is great, thanks for working on this, @atharva-18 !

>  I'm providing the version number and header using IGNITION_GAZEBO_VERSION_HEADER

Perfect

> the hyperlink to the tutorials is specific to Citadel, which might cause an issue in a future version, when the version name changes. Is there a general way to provide the hyperlinks?

How about just linking to https://ignitionrobotics.org/docs/ now?

---

Some UI suggestions that you can feel free to ignore:

* Try using the background color from the theme, `Material.background`, so it should be white for light theme and black for dark
* For the text, try using black or white according to the theme, something like this: `color: Material.theme == Material.Light ? "black" : "white"`
* Give the dialog a maximum width, so that it's a bit more compact.

---

Looking forward to a pull request!

chapulina, 2021-05-06:
Added in #609

closed 2021-05-06 by chapulina