gazebosim/gz-sim #382: Docker images still on bionic, can be updated to focal, nvidia-docker2 is deprecated

russkel, 2020-09-26:
I was trying to get ignition working via docker and I noted that it is still on bionic, which I believe may be interfering with the new nvidia docker support (which doesn't require nvidia-docker2, but nvidia-container-toolkit). 
https://github.com/ignitionrobotics/ign-gazebo/blob/master/docker/Dockerfile.ignition#L7

I am able to get the following to work (please note, the script you include on dockerhub is what I have named `docker-gpu.sh`.
```
./docker-gpu.sh nvidia/opengl:1.2-glvnd-runtime
```
and then installing mesa-utils and running glxgears. I was not able to get glxgears working with your image https://hub.docker.com/r/ignitionrobotics/ignition using the dome tag.

The error messages are your standard:
```
# LIBGL_DEBUG=verbose glxgears
libGL: screen 0 does not appear to be DRI2 capable
libGL: MESA-LOADER: dlopen(/usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL: Can't open configuration file /etc/drirc: No such file or directory.
libGL: Can't open configuration file /root/.drirc: No such file or directory.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  43
  Current serial number in output stream:  42
```
I believe this can be fixed by updating to the focal images.

russkel, 2020-09-26:
Alternatively, you could probably change the base image to `nvidia/opengl:1.2-glvnd-runtime-ubuntu18.04` but given that ROS2 is focal only, it probably makes more sense to get everything on focal.

j-rivero, 2020-09-28:
Hey @russkel: thanks for ticketing the problem. The different branches named `ign-gazeboX` in this repository links with major versions X and every major version supports different platforms. The current `master` branch would be soon `ign-gazebo4` which is part of Ignition Dome and supported on Focal. So we are good to change, would you mind sending the changes against `master` branch in a PR?

russkel, 2020-09-29:
Hi @j-rivero, done as you have requested. I have not yet tried these.

PR #388 (2020-10-06): docker: update Dockerfiles to use focal images
To close #382 

  * use focal base images instead of bionic, including using newer version of libglvnd
  * runtime=nvidia is deprecated, gpus=all has replaced it in newer
  versions of docker

This has not been built or tested by me.
closed 2020-10-06 by russkel