gazebosim/gz-sim #399: Plugin name attributes should be unique

scpeters, 2020-10-08:
I think the `//plugin/@name` attribute is currently being misused by `ign-gazebo`, since it currently contains an `ignition::gazebo::systems::SystemName` and might not be unique among its sibling elements. This is a violation of the SDFormat 1.7 spec, which [requires that sibling elements of any type should have unique names](http://sdformat.org/tutorials?tut=pose_frame_semantics_proposal&cat=pose_semantics_docs&#3-2-unique-names-for-all-sibling-elements). It even violates the convention of earlier versions of the SDFormat spec, which required sibling elements of the same type to have unique names.

I think it would be more suitable to use a custom namespaced attribute to store the system name and give a unique name in the name attribute instead:

~~~
<plugin name='unique_name' filename='...' ignition:system='ignition::gazebo::systems::SystemName' />
~~~

chapulina, 2020-10-08:
>  the //plugin/@name attribute is currently being misused by ign-gazebo

Agreed

> use a custom namespaced attribute

Is this already supported by SDFormat? If so, I suggest we:

* add support for it on Citadel
* convert all our examples to use the new syntax
* deprecate the use of `name` as the class name on Edifice
* remove the use of `name` as the class name on Ign-F

scpeters, 2020-10-12:
> Is this already supported by SDFormat? If so, I suggest we:
> 
> * add support for it on Citadel
> * convert all our examples to use the new syntax
> * deprecate the use of `name` as the class name on Edifice
> * remove the use of `name` as the class name on Ign-F

Well, we currently accept all unrecognized attributes and elements, which includes the namespaced ones; we need to re-enable warnings / errors for unrecognized fields. Some discussion here: https://github.com/osrf/sdformat/issues/327