gazebosim/gz-sim #407: How to consider joint friction in joint controller plugins

diegoferigo, 2020-10-14:
As discussed in https://github.com/ignitionrobotics/ign-physics/issues/124, I am running some experiments with low-level joint torques. I'd like to simulate the effect of joint friction (coulomb and viscous) together with joint controllers.

Ignition Gazebo uses a `JointForceCmd` component to send the joint torque command to the physics engine. Currently, in DART, a joint that receives such command gets configured with `joint->setActuatorType(dart::dynamics::Joint::FORCE)`.

This way, **`JointForceCmd` completely bypasses any modelled joint friction**. It acts as there would be a perfect torque controller, that makes sense in many use-cases.

I don't know if this is the expected behaviour of the existing joint controllers ([`JointController.cc`](https://github.com/ignitionrobotics/ign-gazebo/blob/main/src/systems/joint_controller/JointController.cc#L185-L202) and [`JointPositionController.cc`](https://github.com/ignitionrobotics/ign-gazebo/blob/80cbc2ee0885cfa4ecebbca301916f5f74b8a4ca/src/systems/joint_position_controller/JointPositionController.cc#L231-L241)). Maybe in these two cases what's important is tracking the force / position / velocity  references, and taking friction into account doesn't matter.

However, often it's desirable being able to set a joint force reference that, in presence of friction, won't be followed exactly. High-level controllers (model-base, whole-body, ...) should be able to decide how to deal with friction, maybe adding a feed-forward term (that could not match the real coefficients) in case they want to compensate it. I don't know what is the best way to achieve this behaviour, unless somewhere in the stack there's a hardcoded subtraction before setting the `JointForceCmd`.

Does the Ignition Gazebo behaviour match Gazebo Classic? I don't have much experience on this. Maybe related to https://github.com/osrf/gazebo/issues/1321.

diegoferigo, 2020-10-21:
The change of actuator type introduced in https://github.com/ignitionrobotics/ign-physics/pull/74 might be relevant

azeey, 2023-03-13:
@diegoferigo can this be closed now that we have joint force/torque sensor implemented?

diegoferigo, 2023-03-13:
@azeey it is a while I'm focusing on alternative tools for large-scale simulations, I still haven't tested the merged implementation. From my side, we can close this issue. I can reopen it in case of problems. Thanks @azeey for your work on this.

closed 2023-03-13 by diegoferigo