gazebosim/gz-sim #418: Trajectory animation for non-skeletal actors

chapulina, 2020-10-16:
On Gazebo classic, it is possible to script the trajectory of any geometry using the `<actor>` tag. On Ignition, so far this feature is available only to meshes containing skeletal animations.

On Gazebo classic, that's done by adding any arbitrary `<link>` to an `<actor>`. See [animated_box.world](https://github.com/osrf/gazebo/blob/gazebo11/worlds/animated_box.world) and the [actor tutorial](http://gazebosim.org/tutorials?tut=actor&cat=build_robot). Loading that world on Ignition fails because an animated mesh is expected.

mingfeisun, 2022-02-03:
@chapulina Hi Louise, if this issue has not been addressed yet, I'd like to work out a solution for it.

chapulina, 2022-03-01:
Hi, @mingfeisun ! We're not working on this, so contributions are welcome!

mr3screen, 2026-11-10:
I believe this was also later raised in #1932 and got implemented with https://github.com/gazebosim/gz-sim/pull/1947.
Running the `animated_box.world` with jetty I see the moving box. The only open issue is the appearing error message complaining about the default `<actor><skin><filename>`, which should not be an error anymore, because the example uses a visual and doesn't need skin.
```
(2026-11-10 11:57:27.256) [error] [MeshManager.cc:145] [GUI] Invalid mesh filename extension[/tmp/__default__]
```