gazebosim/gz-sim #420: Migrate Elevator Plugin

nlamprian, 2020-10-18:
Since the list of missing plugins is still long, I thought I would take on migrating the Elevator plugin. Are there any new requirements for it?

chapulina, 2020-10-18:
Thanks!

One difference I can think of is that now we can use an `ignition::msgs::Int32` message instead of using a string message and converting to an int.

Also, you can probably leverage the `JointPositionController` system to do the PID control.

nlamprian, 2020-11-01:
I started by creating a new, more feature-rich, model.

I configured all non-moving links as kinematic. It behaves as expected in Gazebo 9, but not in Ignition Dome.

[image]

The model can be found [here](https://app.ignitionrobotics.org/nlamprian/fuel/models/Elevator).

@chapulina is this a bug, or is the `kinematic` property not implemented yet?

chapulina, 2020-11-02:
> is this a bug, or is the kinematic property not implemented yet?

The kinematic flag isn't supported yet, I ticketed an issue: https://github.com/ignitionrobotics/ign-gazebo/issues/437

commit 7b4fcda6 (2022-02-05): Add elevator system (#535)

* Add elevator system

Closes #420

* Limit line length for transition table

* Add PRIVATE_INCLUDE_DIRS argument to gz_add_system

* Add afsm and metapushkin libraries

* Add namespace to events, actions and guards

* Migrate to afsm

* Fixes
closed 2022-02-05 by nlamprian