gazebosim/gz-sim #428: Support configuring video recording parameters

iche033, 2020-10-26:
The video recording feature uses the default video settings defined in ign-common's [VideoRecorder.cc](https://github.com/ignitionrobotics/ign-common/blob/main/av/src/VideoEncoder.cc) class, i.e. 25 FPS and a max bitrate of 2Mbps. It would be nice to let users to specify their own desired video quality configurations, e.g. through the video recorder service call and/or the GUI widget in the Video Recorder gui plugin.

iche033, 2020-10-26:
related issue: ignitionrobotics/ign-common#109

peci1, 2024-09-16:
This is now supported: https://gazebosim.org/api/sim/9/videorecorder.html#autotoc_md413 .

iche033, 2024-09-16:
https://github.com/gazebosim/gz-sim/pull/864 added support for configuring video recorder bitrate

closed 2024-09-16 by iche033