gazebosim/gz-sim #434: Add more tests to the optical tactile sensor plugin

mcres, 2020-10-30:
Most, if not all, of the [plugin parameters](https://github.com/ignitionrobotics/ign-gazebo/blob/optical_tactile_plugin/tutorials/optical_tactile_plugin.md#parameters) should be tested.

The plugin currently [outputs](https://github.com/ignitionrobotics/ign-gazebo/blob/optical_tactile_plugin/tutorials/optical_tactile_plugin.md#topics--services) an `ignition::msgs::Image` topic. The tests could verify specific sets of parameters and check that.

However, given that most of the parameters have to do with publishing different types&numbers of markers, complicated computations with the previous topic can be avoided if we could access the `/marker_list` topic in the test, which is currently not supported.

mcres, 2020-10-30:
> if we could access the /marker_list topic in the tests

Should I make another issue about this?

chapulina, 2020-11-05:
> Should I make another issue about this?

That sounds like a separate issue, so I think a new ticket would be good.