gazebosim/gz-sim #465: Odometry for omnidirectional robot

FabianPG11, 2020-11-23:
Hi,  I am using ignition gazebo: citadel version to simulate an omnidirectional robot (3 wheels). 

Using gazebo I can control the robot with the plugin libgazebo_ros_planar_move which allows me to define the command and odometry topics, however this plugin is not available on ignition gazebo. 
Is there any similar plugin I could use?. Currently I can make the robot to move using the ignition-gazebo-velocity-control-system plugin, but I am not able to get odometry out of it. The only plugin I have seen with odometry output is diff_drive_system which is not my case.

I appreciate any help.

chapulina, 2020-11-23:
I believe you can get the same control functionality with [ignition::gazebo::systems::VelocityControl](https://github.com/ignitionrobotics/ign-gazebo/blob/a5abacfa74db6e8db0218bd32c55980e15cf249b/examples/worlds/velocity_control.sdf#L309-L312). It accepts a twist message that you can bridge to ROS using `ros_ign_bridge`.

That plugin isn't publishing odometry though.

FabianPG11, 2020-11-23:
Yes, that's how I'm doing it, but the problem is with the odometry. I don't know how to get that.

chapulina, 2020-11-23:
There isn't anything publishing odometry besides the diff drive. I think it would be nice to add a new system that publishes odometry to a topic.

PR #547 (2021-04-21): Add odometry publisher system
Adds a new system that publishes odometry messages for an entity, in a manner similar to what the `ros_planar_move` plugin for Gazebo does. It assumes the robot is on a flat plane and publishes 2D position and velocity data.

There may be other requirements for such a system that I might have missed out on adding, so I can make any changes or additions if necessary.

Closes #465