gazebosim/gz-sim #476: Gazebo crashes when sending remove and create service requests soon after one another

mrushyendra, 2020-12-06:
**Description**
I'm currently creating a GUI plugin to edit lights. As part of the plugin, there are times where we send a series of create and remove service requests to spawn and delete the same light in quick succession to update it (e.g. when the user enters a new pose or direction in the GUI). However, whenever these requests are sent very quickly after one another, Gazebo often crashes with the following backtrace:

```
[ign-1] [GUI] [Err] [BaseStorage.hh:927] Another item already exists with name: {light_name}
[ign-1] /home/rushy/Desktop/ws_dome/install/lib/ruby/ignition/cmdgazebo4.rb:395: [BUG] Segmentation fault at 0x0000000000000000
[ign-1] ruby 2.7.0p0 (2019-12-25 revision 647ee6f091) [x86_64-linux-gnu]
[ign-1] 
[ign-1] -- Control frame information -----------------------------------------------
...
[ign-1] -- C level backtrace information -------------------------------------------
[ign-1] /usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7f64820140a9) [0x7f64820140a9]
[ign-1] /usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7f64820142dc) [0x7f64820142dc]
[ign-1] /usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7f6481e8e24b) [0x7f6481e8e24b]
[ign-1] /usr/lib/x86_64-linux-gnu/libruby-2.7.so.2.7(0x7f6481f916bd) [0x7f6481f916bd]
[ign-1] /lib/x86_64-linux-gnu/libc.so.6(0x7f6481c3b210) [0x7f6481c3b210]
[ign-1] /home/rushy/Desktop/ws_dome/install/lib/libignition-gazebo4-rendering.so.4(0x7f646c15a233) [0x7f646c15a233]
[ign-1] /home/rushy/Desktop/ws_dome/install/lib/libignition-gazebo4-rendering.so.4(_ZN8ignition6gazebo2v412SceneManager11CreateLightEmRKN3sdf3v105LightEm) /home/rushy/Desktop/ws_dome/src/ign-gazebo/src/rendering/SceneManager.cc:859
[ign-1] [0x7f646c111912]
[ign-1] /home/rushy/Desktop/ws_dome/install/lib/ign-gazebo-4/plugins/gui/libGzScene3D.so(_ZN8ignition6gazebo2v411IgnRenderer6RenderEv+0x12b) [0x7f646c1d76db] /home/rushy/Desktop/ws_dome/src/ign-gazebo/src/gui/plugins/scene3d/Scene3D.cc:409
[ign-1] [0x7f646c1d8afc]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN7QObject5eventEP6QEvent+0x23a) [0x7f647d53cd5a]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x86) [0x7f647cad1a66]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x330) [0x7f647cadb0f0]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication15notifyInternal2EP7QObjectP6QEvent+0x18a) [0x7f647d51093a]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN23QCoreApplicationPrivate16sendPostedEventsEP7QObjectiP11QThreadData+0x148) [0x7f647d5135b8]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(0x7f647d568f67) [0x7f647d568f67]
[ign-1] /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x27d) [0x7f647add2fbd]
[ign-1] /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(0x7f647add3240) [0x7f647add3240]
[ign-1] /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7f647add32e3]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x65) [0x7f647d568565]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x12b) [0x7f647d50f4db]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN7QThread4execEv+0x65) [0x7f647d347785]
[ign-1] /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(0x7f647d3489d2) [0x7f647d3489d2]
[ign-1] /lib/x86_64-linux-gnu/libpthread.so.0(start_thread+0xd9) [0x7f6481bdb609]
[ign-1] /lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f6481d17293] ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
[ign-1] 
[ign-1] -- Other runtime information -----------------------------------------------
[ign-1] 
[ign-1] * Loaded script: /home/rushy/Desktop/ws_dome/install/bin/ign
[ign-1] 
...
```
**Example**
I've attached a small reproducible example here:
[create_remove_service_bug.zip](https://github.com/ignitionrobotics/ign-gazebo/files/5648497/create_remove_service_bug.zip)

To run:
1. build the package
2. Launch the `LightTuningLaunch.xml` file
3. Select `Light Tuning` from the dropdown list of plugins on the top right
4. Enter a random light name and select `Add Light`
5. The code in the main `update` loop should either send a create or remove service request in each iteration of the loop. The program should crash soon.

I did a little bit of digging around the Ignition source code, and it seems to be because in this function:
 https://github.com/ignitionrobotics/ign-gazebo/blob/a5abacfa74db6e8db0218bd32c55980e15cf249b/src/rendering/SceneManager.cc#L880

The light returned is a `nullptr`, causing it to crash when `light->SetLocalPose(..)` is called soon after. `CreatePointLight` returns a `nullptr` because it calls `RegisterLight()`, which returns false because it sees a light with the same name already. Note that this happens even after the remove service succeeds and the callback function is called.

Currently we're working around this by just spacing out our service requests, so this issue isn't really a big deal or pressing at all. I'm also guessing the create/remove services were probably not intended for this purpose, so perhaps I'm sort of 'abusing' them. Just posting this issue here in case someone encounters this in the future.

**System Details**
OS: Ubuntu 20.04
Ignition Version: Dome

chapulina, 2020-12-07:
Thank you for the issue, this does sound like a bug.

> As part of the plugin, there are times where we send a series of create and remove service requests to spawn and delete the same light in quick succession to update it

Just curious whether you've looked into calling the [light API](https://ignitionrobotics.org/api/rendering/4.1/classignition_1_1rendering_1_1Light.html) to update it instead of creating a new light every time. I think that would be the ideal implementation, but I'm not sure if it currently has the desired effect.

iche033, 2020-12-08:
We recently added a task for providing better APIs for configuring light properties (and visibility), either through ign topic or ign-gazebo components. Stay tuned...

But yes we should still fix this bug.

mrushyendra, 2020-12-08:
> Just curious whether you've looked into calling the light API to update it instead of creating a new light every time. I think that would be the ideal implementation, but I'm not sure if it currently has the desired effect.

I looked at it earlier, but I wasn't sure if there was a way to get the `Scene` and consequently the relevant `LightPtr` from within a GUI System plugin in order to use the API ?

chapulina, 2020-12-08:
> I wasn't sure if there was a way to get the Scene and consequently the relevant LightPtr from within a GUI System plugin

We've been taking advantage of `ign-rendering`'s singletons for this. See an example here on how to get the scene:

https://github.com/ignitionrobotics/ign-gazebo/blob/a5abacfa74db6e8db0218bd32c55980e15cf249b/src/gui/plugins/grid_config/GridConfig.cc#L152-L187

Just be sure to only edit the scene within the render event:

https://github.com/ignitionrobotics/ign-gazebo/blob/a5abacfa74db6e8db0218bd32c55980e15cf249b/src/gui/plugins/grid_config/GridConfig.cc#L88-L95

mrushyendra, 2020-12-08:
This works great for our use case! Thanks

ahcorde, 2021-01-13:
This issue is resolved. Feel free to reopen it if there is still some problems

closed 2021-01-13 by ahcorde