gazebosim/gz-sim #489: Configure physics properties

iche033, 2020-12-15:
Support configuring physics properties like max step size, real time update rate, gravity, etc (see [Physics](https://github.com/osrf/sdformat/blob/master/src/Physics.cc) SDF DOM and perhaps also some of the fields in [physics.proto](https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs6/proto/ignition/msgs/physics.proto)) through ign gazebo components and/or ign-transport

Additionally a component inspector to let users edit these properties through the gui

mrushyendra, 2020-12-18:
I made some initial changes [here](https://github.com/ignitionrobotics/ign-gazebo/compare/ign-gazebo4...mrushyendra:feature/configure_physics_properties#) to add a component that allows users to change the step size or target rtf. While it works in practice, I was wondering where the best place to actually update the `SimulationRunner`'s `stepSize` and `desiredRtf` properties would be. Currently I'm doing it in `step` here:

https://github.com/mrushyendra/ign-gazebo/blob/56f800e073db107120f924f196fcb92e818b38f6/src/SimulationRunner.cc#L740

but was wondering if that is the right approach. Thank you!

ahcorde, 2020-12-18:
Hi @mrushyendra 

Thank you for picking this issue. I think it would be great if you can create a PR because we can discuss better the code and see easily the new code.

I think you can also consider two more things:

 - It would also be great to make the `component inspector` call those new topics to edit physics from the GUI.
 - the PhysicCmd should probably be handled by the `UserCommands` system.

This PR https://github.com/ignitionrobotics/ign-gazebo/pull/482 is adding light control, you can get some inspiration.

luca-della-vedova, 2021-01-06:
I started to build on top of the work by @mrushyendra and should be ready to open a draft PR for component inspector control of physics parameters soon.

closed 2021-02-02 by chapulina