gazebosim/gz-sim #496: Transform Controller Markers Behind Environment Models

ibrahimkakbar, 2020-12-17:
The GUI Transform Control displays of a robot model for the linear/rotational transformations are behind the world environment and only appear on the robot model. I have attached images of:

-  a regular version which shows some clipping on the rotation circles
- a zoomed in version to show the complete rotation circles displayed on the robot model
- a zoomed out version to show the bands disappear and only be visible in open space
[image]
[image]
[image]

chapulina, 2021-01-11:
Thanks for the issue. There's probably some property of the cave materials that is making it be rendered above the GUI tools. The fix may end up being in `ign-rendering`, but it's fine to keep tracking this here.

angelacmaio, 2021-03-03:
Seconding this one. It's quite difficult to build worlds with the GUI when you can't count on the arrows. I did not see this issue with any materials displaying over the arrows until the switch from Blueprint to Dome.

chapulina, 2021-03-05:
I did some tests and it looks like the markers, and probably anything with `IGN_VISIBILITY_GUI` don't show through any PBR materials. It's easy to reproduce with [this model](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Rescue%20Randy):

[image]

@iche033 , any guidance on how to fix this?

iche033, 2021-03-06:
here's a fix in https://github.com/ignitionrobotics/ign-rendering/pull/266

closed 2021-03-13 by iche033