gazebosim/gz-sim #504: Disable gravity for links

chapulina, 2020-12-18:
Gazebo Classic supports the [`<link><gravity>`](http://sdformat.org/spec?ver=1.7&elem=link#link_gravity) SDF element, which makes links float.

That hasn't been ported to Ignition yet, we aren't even parsing the flag into [sdf::Link](https://github.com/osrf/sdformat/blob/master/include/sdf/Link.hh).

slim71, 2024-06-05:
Any update on this? Specifically, in my case, into Garden...

iche033, 2024-06-06:
the sdformat PRs were merged and forward ported to garden and newer versions. We'll need to make sdformat releases so that the gz-physics and gz-sim PRs can be merged.

slim71, 2024-06-06:
not to rush or anything, but does this have an ETA? is it scheduled?

iche033, 2024-06-06:
I'll kick off the sdformat releases, starting with fortress https://github.com/gazebosim/sdformat/pull/1430