gazebosim/gz-sim #509: Move video recording logic from Scene3D to VideoRecorder

chapulina, 2020-12-22:
The [video recorder plugin ](https://github.com/ignitionrobotics/ign-gazebo/tree/ign-gazebo4/src/gui/plugins/video_recorder) provides a button for recording videos. However, all the logic involved in actually recording is contained [within the scene 3D plugin](https://github.com/ignitionrobotics/ign-gazebo/blob/efe8ea15e304748c195b46f8697033682cc05d45/src/gui/plugins/scene3d/Scene3D.cc#L477). All the video recorder plugin does is calling a service to start and stop recording.

This is not a very good separation of concerns. I believe all the video rendering and synchronizing logic that is currently in `Scene3D` could be moved to the `VideoRecorder`, making use of the [render event](https://github.com/ignitionrobotics/ign-gazebo/blob/efe8ea15e304748c195b46f8697033682cc05d45/include/ignition/gazebo/gui/GuiEvents.hh#L152-L162). @iche033 , do you think that should work?

The scene 3D code is already getting quite bloated. We should try to keep logic that is optional, and not directly necessary to render the 3D scene, inside each specific plugin. That should make plugins easier to maintain and users can easily opt-out of them.

iche033, 2020-12-22:
I think it's possible. We would need to be able to access the gui camera though. I think we'll just have to make some assumption that the scene on the gui side only has one camera (which should be the case for now) and look for that camera from the video recorder plugin through the render engine singleton.

chapulina, 2021-12-13:
This has been done on Fortress: https://github.com/ignitionrobotics/ign-gazebo/pull/900

closed 2021-12-13 by chapulina