gazebosim/gz-sim #510: Can't visualise lidar and use ros_ign_bridge at the same time.

DrFKNoble, 2020-12-22:
Hi there,

I'm trying to use Ignition Gazebo's Visualize Lidar plugin and ros_ign_bridge at the same time; however, whenever I do, I get the following error:

```console
[GUI] [Err] [VisualizeLidar.cc:275] Error entity model_with_lidar/link/gpu_lidar doesn't exist and cannot be used to set lidar visual pose
```console

The command I use to bridge between Ignition Gazebo and ROS is as follows:
```console
ros2 run ros_ign_bridge parameter_bridge /lidar@sensor_msgs/msg/LaserScan@ignition.msgs.LaserScan
```

If I use either on their own, they work fine; I can use the plugin as intended and can use rqt to check the data being received by ROS.

I am using Ignition Gazebo, version 4.1.0 and have tested this with the example world provided here: https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/examples/worlds/visualize_lidar.sdf. 

Any help getting this to work would be greatly appreciated.

Best regards,

Frazer

chapulina, 2020-12-23:
Interesting, it seems that the bi-directional bridge may be republishing its own messages. I don't see the error if I use a uni-directional bridge from ign to ROS:

`ros2 run ros_ign_bridge parameter_bridge /lidar@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan`

DrFKNoble, 2020-12-23:
@chapulina Thanks for your reply! I hadn't even thought to try the uni-directional bridge to send the sensor data. I just tried and it worked as you said. That resolves my issue.