gazebosim/gz-sim #520: make doc fails with segmentation fault in doxygen 1.9.0 on macOS

scpeters, 2020-12-30:
Since doxygen 1.9.0 was released into homebrew-core in https://github.com/Homebrew/homebrew-core/pull/67817, `make doc` has been failing for ign-gazebo with a segmentation fault in `doxygen` itself:

* [image] https://build.osrfoundation.org/view/os_homebrew/job/ignition_gazebo-ci-ign-gazebo4-homebrew-amd64/13/

~~~
make[2]: *** [CMakeFiles/doc] Segmentation fault: 11
~~~

Running `make doc VERBOSE=1` shows that the following command is used: `doxygen build/api_tagfile.dox`. By editing `build/api_tagfile.dox` so that `QUIET = NO`, you can get verbose output from `make doc`. I see the following lines just before the failure:

~~~
Preprocessing <prefix>/ign-gazebo/include/ignition/gazebo/Util.hh...
Parsing file <prefix>/ign-gazebo/include/ignition/gazebo/Util.hh...
Building macro definition list...
Building group list...
Building directory list...
Building namespace list...
Segmentation fault: 11
~~~

I've built doxygen from source in debug mode and have isolated the segmentation fault to [a specific line in doxygen.cpp](https://github.com/doxygen/doxygen/blob/Release_1_9_0/src/doxygen.cpp#L1569). I will submit a bug report to doxygen. In the meantime, we may want to disable generation of docs in this repository.

scpeters, 2020-12-30:
I uncommented a printf from doxygen and get the following output:

~~~
diff --git a/src/doxygen.cpp b/src/doxygen.cpp
index 3472f8f1..d4546ff3 100644
--- a/src/doxygen.cpp
+++ b/src/doxygen.cpp
@@ -1558,7 +1558,7 @@ static void buildNamespaceList(const Entry *root)
           tagName     = tagInfo->tagName;
           tagFileName = tagInfo->fileName;
         }
-        //printf("++ new namespace %s lang=%s tagName=%s\n",fullName.data(),langToString(root->lang).data(),tagName.data());
+        printf("++ new namespace %s lang=%s tagName=%s\n",fullName.data(),langToString(root->lang).data(),tagName.data());
         // add namespace to the list
         nd = toNamespaceDefMutable(
             Doxygen::namespaceLinkedMap->add(fullName,
~~~

~~~
Building macro definition list...
Building group list...
Building directory list...
Building namespace list...
++ new namespace std lang=Unknown tagName=/usr/local/share/ignition/ignition-cmake2/doxygen/cppreference.tag.xml
++ new namespace std::regex_constants lang=Unknown tagName=/usr/local/share/ignition/ignition-cmake2/doxygen/cppreference.tag.xml
++ new namespace std::chrono lang=Unknown tagName=/usr/local/share/ignition/ignition-cmake2/doxygen/cppreference.tag.xml
++ new namespace std::rel_ops lang=Unknown tagName=/usr/local/share/ignition/ignition-cmake2/doxygen/cppreference.tag.xml
++ new namespace std::this_thread lang=Unknown tagName=/usr/local/share/ignition/ignition-cmake2/doxygen/cppreference.tag.xml
++ new namespace std::experimental lang=Unknown tagName=/usr/local/share/ignition/ignition-cmake2/doxygen/cppreference.tag.xml
++ new namespace ignition lang=Unknown tagName=/usr/local/Cellar/ignition-math6/6.8.0~pre1/share/ignition/ignition-math6/ignition-math6.tag.xml
++ new namespace ignition::math lang=Unknown tagName=/usr/local/Cellar/ignition-math6/6.8.0~pre1/share/ignition/ignition-math6/ignition-math6.tag.xml
++ new namespace ignition::math::eigen3 lang=Unknown tagName=/usr/local/Cellar/ignition-math6/6.8.0~pre1/share/ignition/ignition-math6/ignition-math6.tag.xml
++ new namespace ignition::math::IGNITION_MATH_VERSION_NAMESPACE lang=Unknown tagName=/usr/local/Cellar/ignition-math6/6.8.0~pre1/share/ignition/ignition-math6/ignition-math6.tag.xml
++ new namespace ignition::math::IGNITION_MATH_VERSION_NAMESPACE::graph lang=Unknown tagName=/usr/local/Cellar/ignition-math6/6.8.0~pre1/share/ignition/ignition-math6/ignition-math6.tag.xml
++ new namespace ignition::msgs lang=Unknown tagName=/usr/local/Cellar/ignition-msgs4/4.9.0_5/share/ignition/ignition-msgs4/ignition-msgs4.tag.xml
++ new namespace ignition::msgs::IGNITION_MSGS_VERSION_NAMESPACE lang=Unknown tagName=/usr/local/Cellar/ignition-msgs4/4.9.0_5/share/ignition/ignition-msgs4/ignition-msgs4.tag.xml
++ new namespace ignition::physics lang=Unknown tagName=/usr/local/Cellar/ignition-physics1/1.10.0_2/share/ignition/ignition-physics1/ignition-physics1.tag.xml
++ new namespace ignition::physics::dartsim lang=Unknown tagName=/usr/local/Cellar/ignition-physics1/1.10.0_2/share/ignition/ignition-physics1/ignition-physics1.tag.xml
++ new namespace ignition::physics::mesh lang=Unknown tagName=/usr/local/Cellar/ignition-physics1/1.10.0_2/share/ignition/ignition-physics1/ignition-physics1.tag.xml
++ new namespace ignition::physics::sdf lang=Unknown tagName=/usr/local/Cellar/ignition-physics1/1.10.0_2/share/ignition/ignition-physics1/ignition-physics1.tag.xml
++ new namespace ignition::sensors lang=Unknown tagName=/usr/local/Cellar/ignition-sensors2/2.9.1/share/ignition/ignition-sensors2/ignition-sensors2.tag.xml
++ new namespace ignition::sensors::IGNITION_SENSORS_VERSION_NAMESPACE lang=Unknown tagName=/usr/local/Cellar/ignition-sensors2/2.9.1/share/ignition/ignition-sensors2/ignition-sensors2.tag.xml
++ new namespace ignition::gazebo lang=C++ tagName=(null)
++ new namespace ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE lang=C++ tagName=(null)
++ new namespace ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::components lang=C++ tagName=(null)
++ new namespace ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::serializers lang=C++ tagName=(null)
++ new namespace ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::traits lang=C++ tagName=(null)
++ new namespace ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::events lang=C++ tagName=(null)
++ new namespace ignition::gazebo::IGNITION_GAZEBO_VERSION_NAMESPACE::gui lang=C++ tagName=(null)
++ new namespace ignition::gazebo::gui lang=C++ tagName=(null)
Process 23471 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000100389654 doxygen`buildNamespaceList(root=0x00000001032ce218) at doxygen.cpp:1569:13
   1566	                createNamespaceDef(tagInfo?tagName:root->fileName,root->startLine,
   1567	                             root->startColumn,fullName,tagName,tagFileName,
   1568	                             root->type,root->spec&Entry::Published))));
-> 1569	        nd->setDocumentation(root->doc,root->docFile,root->docLine); // copy docs to definition
   1570	        nd->setBriefDescription(root->brief,root->briefFile,root->briefLine);
   1571	        nd->addSectionsToDefinition(root->anchors);
   1572	        nd->setHidden(root->hidden);
Target 0: (doxygen) stopped.
(lldb) print nd
(NamespaceDefMutable *) $0 = 0x0000000000000000
~~~

it seems the output of `createNamespaceDef` on [line 1563](https://github.com/doxygen/doxygen/blob/Release_1_9_0/src/doxygen.cpp#L1563) is a `nullptr`

scpeters, 2020-12-30:
it seems similar to https://github.com/doxygen/doxygen/issues/8192 but not quite the same

scpeters, 2020-12-30:
I believe the following patch to ign-gazebo fixes it:

~~~
diff --git a/include/ignition/gazebo/gui/GuiEvents.hh b/include/ignition/gazebo/gui/GuiEvents.hh
index 524805dd..90efc184 100644
--- a/include/ignition/gazebo/gui/GuiEvents.hh
+++ b/include/ignition/gazebo/gui/GuiEvents.hh
@@ -29,9 +29,9 @@ namespace ignition
 {
 namespace gazebo
 {
-namespace gui {
 // Inline bracket to help doxygen filtering.
 inline namespace IGNITION_GAZEBO_VERSION_NAMESPACE {
+namespace gui {
 /// \brief Namespace for all events. Refer to the EventManager class for
 /// more information about events.
 namespace events
~~~

that may break ABI, but it could be helpful in creating a small reproduction of the error in order to report a proper issue to doxygen

scpeters, 2020-12-30:
> I believe the following patch to ign-gazebo fixes it:
>
> ```
> diff --git a/include/ignition/gazebo/gui/GuiEvents.hh b/include/ignition/gazebo/gui/GuiEvents.hh
> index 524805dd..90efc184 100644
> --- [...]

that seems to fix it when I run doxygen in debug mode but not with the homebrew binaries (see failed CI in #521)

chapulina, 2020-12-30:
Thanks for debugging, @scpeters 

> we may want to disable generation of docs in this repository.

Another thing we can consider is not generating docs in macOS. Our official docs are deployed using Ubuntu, and I'm not sure if many users are making use of documentation generation on macOS. I'd be ok only supporting doc generation (and other tooling like static codecheck and coverage) only on Linux to reduce our maintenance surface.

That said, I think it's good that we're catching this early and can solve it before Doxygen 1.9.0 shows up on Ubuntu.

scpeters, 2020-12-30:
> Another thing we can consider is not generating docs in macOS

yes, that's the right step to take

> That said, I think it's good that we're catching this early and can solve it before Doxygen 1.9.0 shows up on Ubuntu.

I just tried building doxygen from source on 18.04 with gcc and I can't reproduce it, so I think it is specific to macOS or clang.

ok, now I'm going to enjoy my vacation

chapulina, 2020-12-30:
> not generating docs in macOS

Ok, an attempt here: https://github.com/osrf/homebrew-simulation/pull/1258/

scpeters, 2020-12-30:
> > not generating docs in macOS
> 
> Ok, an attempt here: [osrf/homebrew-simulation#1258](https://github.com/osrf/homebrew-simulation/pull/1258)

that would disable doxygen for all our packages. I think only ign-gazebo is suffering from this issue, so I would just put an `if(NOT APPLE)` block around `ign_create_docs` in CMakeLists.txt for ign-gazebo

chapulina, 2020-12-30:
> that would disable doxygen for all our packages.

The way I understood it, that would still keep doxygen for those specific packages, and downstream packages that want to generate docs would need to declare the build dependency themselves. That sounds like the most correct thing to do, I don't think downstream packages should be relying on a build dependency from their dependencies, right?

I can go ahead and add doxygen as an optional build dependency for all packages downstream of those, except for `ign-gazebo`, if you agree.

doxygen, 2020-12-30:
Can you check if the crash is solved when applying this fix to doxygen? https://github.com/doxygen/doxygen/commit/4a070f8b60eefa7997c9fdd59553970de3645207

chapulina, 2020-12-30:
Here's the suppression to make docs for macOS: https://github.com/ignitionrobotics/ign-gazebo/pull/523

scpeters, 2021-01-01:
> Can you check if the crash is solved when applying this fix to doxygen? [doxygen/doxygen@4a070f8](https://github.com/doxygen/doxygen/commit/4a070f8b60eefa7997c9fdd59553970de3645207)

Yes! Thank you so much; that fixes the segmentation fault when run against our tagfile, but it looks `make doc` has 2 invocations of `doxygen` and the 2nd one has a segmentation fault as well. I am investigating that now to get a backtrace...

scpeters, 2021-01-01:
> > Can you check if the crash is solved when applying this fix to doxygen? [doxygen/doxygen@4a070f8](https://github.com/doxygen/doxygen/commit/4a070f8b60eefa7997c9fdd59553970de3645207)
> 
> Yes! Thank you so much; that fixes the segmentation fault when run against our `api_tagfile.dox`, but it looks `make doc` has 2 invocations of `doxygen` and the 2nd one has a segmentation fault as well. I am investigating that now to get a backtrace...

never mind; it's all fixed by https://github.com/doxygen/doxygen/commit/4a070f8b60eefa7997c9fdd59553970de3645207; thanks!

PR #523 (2021-01-04): Don't make docs on macOS
Resolves #520.
closed 2021-01-04 by chapulina