gazebosim/gz-sim #526: seg fault

acxz, 2021-01-01:
Running `ign gazebo` results in a seg fault:

<details>
<summary> Log </summary>

  ```
/usr/lib/ruby/ignition/cmdgazebo4.rb:426: [BUG] Segmentation fault at 0x0000000000000000
ruby 2.7.2p137 (2020-10-01 revision 5445e04352) [x86_64-linux]

-- Control frame information -----------------------------------------------
c:0007 p:---- s:0058 e:000057 CFUNC  :call
c:0006 p:0016 s:0053 e:000052 METHOD /usr/lib/ruby/ignition/cmdgazebo4.rb:426
c:0005 p:0060 s:0047 e:000046 BLOCK  /usr/lib/ruby/ignition/cmdgazebo4.rb:456 [FINISH]
c:0004 p:---- s:0044 e:000043 CFUNC  :fork
c:0003 p:0654 s:0040 e:000039 METHOD /usr/lib/ruby/ignition/cmdgazebo4.rb:452
c:0002 p:0615 s:0019 E:0007a8 EVAL   /usr/bin/ign:267 [FINISH]
c:0001 p:0000 s:0003 E:0018a0 (none) [FINISH]

-- Ruby level backtrace information ----------------------------------------
/usr/bin/ign:267:in `<main>'
/usr/lib/ruby/ignition/cmdgazebo4.rb:452:in `execute'
/usr/lib/ruby/ignition/cmdgazebo4.rb:452:in `fork'
/usr/lib/ruby/ignition/cmdgazebo4.rb:456:in `block in execute'
/usr/lib/ruby/ignition/cmdgazebo4.rb:426:in `runGui'
/usr/lib/ruby/ignition/cmdgazebo4.rb:426:in `call'

-- Machine register context ------------------------------------------------
 RIP: 0x00007f3cf49c50f5 RBP: 0x00007f3cc5113950 RSP: 0x00007f3cc51138e0
 RAX: 0x0000000000000008 RBX: 0x0000000000000010 RCX: 0x00007f3b63a6d9ca
 RDX: 0x0000557de0fb0ef0 RDI: 0x00007f3c909c07c0 RSI: 0x00007f3c909c1e80
  R8: 0x00007f3c909c1e80  R9: 0x0000000000000005 R10: 0x0000000000000015
 R11: 0x0000000000000000 R12: 0x00007f3c909c07b0 R13: 0x0000000000000000
 R14: 0x00007f3cc5113d00 R15: 0x0000557de040a140 EFL: 0x0000000000010202

-- C level backtrace information -------------------------------------------
/usr/lib/libruby.so.2.7(0x7f3cfa104c04) [0x7f3cfa104c04]
/usr/lib/libruby.so.2.7(0x7f3cf9f6f019) [0x7f3cf9f6f019]
/usr/lib/libruby.so.2.7(0x7f3cfa07fb3a) [0x7f3cfa07fb3a]
/usr/lib/libc.so.6(__restore_rt+0x0) [0x7f3cf9d3f6a0]
/usr/lib/libQt5Quick.so.5(_ZNK12QQuickWindow29createTextureFromNativeObjectENS_16NativeObjectTypeEPKviRK5QSize6QFlagsINS_19CreateTextureOptionEE+0xf5) [0x7f3cf49c50f5]
/usr/lib/ign-gazebo-4/plugins/gui/libGzScene3D.so(_ZN8ignition6gazebo2v411TextureNodeC1EP12QQuickWindow+0xa3) [0x7f3cec477273]
/usr/lib/ign-gazebo-4/plugins/gui/libGzScene3D.so(_ZN8ignition6gazebo2v416RenderWindowItem15updatePaintNodeEP7QSGNodePN10QQuickItem19UpdatePaintNodeDataE+0xb4) [0x7f3cec4779d4]
/usr/lib/libQt5Quick.so.5(_ZN19QQuickWindowPrivate15updateDirtyNodeEP10QQuickItem+0x858) [0x7f3cf49c70f8]
/usr/lib/libQt5Quick.so.5(_ZN19QQuickWindowPrivate16updateDirtyNodesEv+0x8c) [0x7f3cf49c79ac]
/usr/lib/libQt5Quick.so.5(_ZN19QQuickWindowPrivate14syncSceneGraphEv+0xa1) [0x7f3cf49c9311]
/usr/lib/libQt5Quick.so.5(0x7f3cf4968c00) [0x7f3cf4968c00]
/usr/lib/libQt5Quick.so.5(0x7f3cf496ab68) [0x7f3cf496ab68]
/usr/lib/libQt5Quick.so.5(0x7f3cf496b897) [0x7f3cf496b897]
/usr/lib/libQt5Core.so.5(0x7f3cf569af0f) [0x7f3cf569af0f]
/usr/lib/libpthread.so.0(start_thread+0xe9) [0x7f3cf9ccf3e9]
/usr/lib/libc.so.6(clone+0x43) [0x7f3cf9e02293]

-- Other runtime information -----------------------------------------------

* Loaded script: /usr/bin/ign

* Loaded features:

    0 enumerator.so
    1 thread.rb
    2 rational.so
    3 complex.so
    4 ruby2_keywords.rb
    5 /usr/lib/ruby/2.7.0/x86_64-linux/enc/encdb.so
    6 /usr/lib/ruby/2.7.0/x86_64-linux/enc/trans/transdb.so
    7 /usr/lib/ruby/2.7.0/x86_64-linux/rbconfig.rb
    8 /usr/lib/ruby/2.7.0/rubygems/compatibility.rb
    9 /usr/lib/ruby/2.7.0/rubygems/defaults.rb
   10 /usr/lib/ruby/2.7.0/rubygems/deprecate.rb
   11 /usr/lib/ruby/2.7.0/rubygems/errors.rb
[... 1842 lines cut ...]
7f3cf9ecb000-7f3cf9ef8000 r--p 00000000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f3cf9ef8000-7f3cfa10f000 r-xp 0002d000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f3cfa10f000-7f3cfa207000 r--p 00244000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f3cfa207000-7f3cfa210000 r--p 0033b000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f3cfa210000-7f3cfa211000 rw-p 00344000 103:04 6208497                   /usr/lib/libruby.so.2.7.2
7f3cfa211000-7f3cfa22d000 rw-p 00000000 00:00 0 
7f3cfa22d000-7f3cfa22f000 r--p 00000000 103:04 6163799                   /usr/lib/ld-2.32.so
7f3cfa22f000-7f3cfa250000 r-xp 00002000 103:04 6163799                   /usr/lib/ld-2.32.so
7f3cfa250000-7f3cfa259000 r--p 00023000 103:04 6163799                   /usr/lib/ld-2.32.so
7f3cfa259000-7f3cfa25a000 r--p 0002b000 103:04 6163799                   /usr/lib/ld-2.32.so
7f3cfa25a000-7f3cfa25c000 rw-p 0002c000 103:04 6163799                   /usr/lib/ld-2.32.so
7ffc94ab1000-7ffc952b0000 rw-p 00000000 00:00 0                          [stack]
7ffc953c2000-7ffc953c6000 r--p 00000000 00:00 0                          [vvar]
7ffc953c6000-7ffc953c8000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 --xp 00000000 00:00 0                  [vsyscall]
  ```

</details>

Sys info:
OS: Arch Linux
GCC: 10.2.0
Qt: 5.15.2
ruby: 2.7.2
protobuf: 3.12.4

acxz, 2021-01-01:
Believe the following codeblock is the culprint:

https://github.com/ignitionrobotics/ign-gazebo/blob/a830b443989877b1e83170390f483b6c02952b25/src/gui/plugins/scene3d/Scene3D.cc#L1973-L1975

Found the following link which mentions that `nullptr` causes a crash on qt 5.15.2:
https://bugreports.qt.io/browse/QTBUG-89114

ahcorde, 2021-01-04:
A similar issue was resolved in ign-gui https://github.com/ignitionrobotics/ign-gui/commit/19f9dae1edfedeabee61269f7dee326b98a1a162. Do you mind to open a PR ?

ahcorde, 2021-01-04:
Thank you for the PR  https://github.com/ignitionrobotics/ign-gazebo/pull/527/files

PR #527 (2021-01-07): change nullptr to a int ptr for qt 5.15.2 bug
See: https://bugreports.qt.io/browse/QTBUG-89114

Resolves #526
closed 2021-01-07 by chapulina