gazebosim/gz-sim #545: EventManager class can not be exported in MSVC

j-rivero, 2021-01-11:
There is a problem with the use on `std::unordered_map` together with `std::unique_ptr`when enabling symbol visibility in Visual Studio. [Relevant code](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/include/ignition/gazebo/EventManager.hh#L139-L142): 

```c++
  /// \brief Container of used signals.
  private: std::unordered_map<TypeInfoRef,
                              std::unique_ptr<std::string>,
                              Hasher, EqualTo> events;
```

Compiler error:

> C:/data/msvc/14.28.29333/include\list(1284): error C2679: binary '=': no operator found which takes a right-hand operand of type 'const std::pair<const EventManager::TypeInfoRef,std::unique_ptr<std::string,std::default_delete<std::string>>>' (or there is no acceptable conversion)

The problem is gone if: there is no symbol exported (no visibility) or std::unitque_ptr is [removed from that code](https://godbolt.org/z/Pvxfdj) and use plain strings. The problem is only present in Visual Studio, code is fine for clang or gcc.

I've created a minimal code example: https://github.com/j-rivero/msvc_unordered_map

mjcarroll, 2021-01-11:
That's unfortunate.  I think we can easily not export that symbol since it's private anyway?

j-rivero, 2021-01-11:
> That's unfortunate. I think we can easily not export that symbol since it's private anyway?

That's the plan by now. Not exporting the class means that some tests can not work, here is my preliminary list:
```
* INTEGRATION_link
* INTEGRATION_sdf_frame_semantics
* EventManager_TEST
* PeerTracker_TEST
* SdfGenerator_TEST
```

chapulina, 2021-01-12:
> Not exporting the class means that some tests can not work

I imagine user code that links against that may not compile either right? That's unfortunate.

> we can easily not export that symbol since it's private anyway?

That would break ABI, right?

mjcarroll, 2021-01-12:
We can export everything except for that one private symbol, no?

j-rivero, 2021-01-13:
I think I've found a workaround. Remove the `.cc` file and make the class header only ([see commit](https://github.com/ignitionrobotics/ign-gazebo/commit/6a595c11bc52a7f138328d0dd720834a2515cb3f)). That way all the code will go into the compilations units via header (for tests or third party software) and no linking should be needed so we don't need the visibility. That should not impact on the ABI since the symbols are being generated anyway but they come from the compilations instead of resolving them via linking.

chapulina, 2021-01-13:
Interesting, I wonder why we had that CC file in the first place.

j-rivero, 2021-02-01:
The workaround seems to be working the whole CI is ready. Closing.

closed 2021-02-01 by j-rivero