gazebosim/gz-sim #550: [GUI] [Err] [PathManager.cc:57] Service call timed out for [/gazebo/resource_paths/get]

ekorudiawan, 2021-01-13:
I installed Ignition-Citadel and try running the ignition from [get started tutorial](https://ignitionrobotics.org/docs/all/getstarted) using this command

```bash
ign gazebo shapes.sdf
```

The GUI is not showing anything and I got an error message

```bash
[GUI] [Err] [PathManager.cc:57] Service call timed out for [/gazebo/resource_paths/get]
```

I try to add Environment variables manually

```bash
export IGN_GUI_PLUGIN_PATH=/usr/share/ignition/ignition-gui3
export IGN_GAZEBO_SYSTEM_PLUGIN_PATH=/usr/share/ignition/ignition-gazebo3
export IGN_GAZEBO_RESOURCE_PATH=/usr/share/ignition/ignition-gazebo3/worlds
```

but it still can't resolve the issue

ahcorde, 2021-01-13:
Did you have another ign-gazebo version installed ? Try to remove the Gazebo gui config file.

```bash
rm -r ~/.ignition/gazebo/gui.config
```

You can also run Gazebo with higher log level.

```bash
ign gazebo -v 4 shapes.sdf
```

ekorudiawan, 2021-01-13:
I checked this file `~/.ignition/gazebo/gui.config` and it does not appear on my computer and I don't have another ignition. But previously, I installed ignition dome and got the same problem. I completely uninstall the dome version and install the LTS version, but the problem still appears.

This is the log output after running with `ign gazebo -v 4 shapes.sdf

```bash
[Msg] Ignition Gazebo Server v3.6.0
[Msg] Ignition Gazebo GUI    v3.6.0
[Msg] Loading SDF world file[/usr/share/ignition/ignition-gazebo3/worlds/shapes.sdf].
qt5ct: using qt5ct plugin
[Dbg] [Application.cc:87] Initializing application.
[GUI] [Dbg] [Application.cc:407] Create main window
[Dbg] [EntityComponentManager.cc:596] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
[Dbg] [EntityComponentManager.cc:596] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
[Dbg] [EntityComponentManager.cc:596] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
[Dbg] [EntityComponentManager.cc:596] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
[Dbg] [EntityComponentManager.cc:596] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
[Dbg] [EntityComponentManager.cc:596] Using components of type [17459188283658606303] / [ign_gazebo_components.RenderEngineServerPlugin].
[Dbg] [EntityComponentManager.cc:596] Using components of type [8705992680619689917] / [ign_gazebo_components.RenderEngineGuiPlugin].
[Dbg] [EntityComponentManager.cc:596] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
[Dbg] [EntityComponentManager.cc:596] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
[Dbg] [EntityComponentManager.cc:596] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
[Dbg] [EntityComponentManager.cc:596] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
[Dbg] [EntityComponentManager.cc:596] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
[Dbg] [EntityComponentManager.cc:596] Using components of type [17605309075052480649] / [ign_gazebo_components.Atmosphere].
[Dbg] [EntityComponentManager.cc:596] Using components of type [8064491505919932473] / [ign_gazebo_components.Level].
[Dbg] [EntityComponentManager.cc:596] Using components of type [2668898242563798256] / [ign_gazebo_components.DefaultLevel].
[Dbg] [EntityComponentManager.cc:596] Using components of type [11371360182141354106] / [ign_gazebo_components.LevelEntityNames].
[Dbg] [Physics.cc:600] Loaded [ignition::physics::dartsim::Plugin] from library [/usr/lib/x86_64-linux-gnu/ign-physics-2/engine-plugins/libignition-physics-dartsim-plugin.so]
[Dbg] [SimulationRunner.cc:764] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
[Msg] Create service on [/world/shapes/create]
[Msg] Remove service on [/world/shapes/remove]
[Msg] Pose service on [/world/shapes/set_pose]
[Dbg] [SimulationRunner.cc:764] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
[Dbg] [SimulationRunner.cc:764] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
[Dbg] [EntityComponentManager.cc:596] Using components of type [4981278897826323946] / [ign_gazebo_components.WorldSdf].
[Dbg] [EntityComponentManager.cc:596] Using components of type [6687176221774458630] / [ign_gazebo_components.Model].
[Dbg] [EntityComponentManager.cc:596] Using components of type [6612894081701502240] / [ign_gazebo_components.Pose].
[Dbg] [EntityComponentManager.cc:596] Using components of type [8546580419506082455] / [ign_gazebo_components.Static].
[Dbg] [EntityComponentManager.cc:596] Using components of type [9712747055438129860] / [ign_gazebo_components.WindMode].
[Dbg] [EntityComponentManager.cc:596] Using components of type [5661073481138181711] / [ign_gazebo_components.SelfCollide].
[Dbg] [EntityComponentManager.cc:596] Using components of type [11683062252779233161] / [ign_gazebo_components.SourceFilePath].
[Dbg] [EntityComponentManager.cc:596] Using components of type [5081358965268446661] / [ign_gazebo_components.Link].
[Dbg] [EntityComponentManager.cc:596] Using components of type [8112400427272910195] / [ign_gazebo_components.Inertial].
[Dbg] [EntityComponentManager.cc:596] Using components of type [16454635107327670381] / [ign_gazebo_components.Visual].
[Dbg] [EntityComponentManager.cc:596] Using components of type [13011964647677164955] / [ign_gazebo_components.CastShadows].
[Dbg] [EntityComponentManager.cc:596] Using components of type [13440282432131634483] / [ign_gazebo_components.Transparency].
[Dbg] [EntityComponentManager.cc:596] Using components of type [17121648710877364109] / [ign_gazebo_components.Geometry].
[Dbg] [EntityComponentManager.cc:596] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
[Dbg] [EntityComponentManager.cc:596] Using components of type [17938588655714334139] / [ign_gazebo_components.Collision].
[Dbg] [EntityComponentManager.cc:596] Using components of type [9225962031573086509] / [ign_gazebo_components.CollisionElement].
[Dbg] [EntityComponentManager.cc:596] Using components of type [10522242218202596205] / [ign_gazebo_components.CanonicalLink].
[... cut ...]
[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: The following application identifiers have not been set: QVector("organizationName", "organizationDomain")
[GUI] [Dbg] [Gui.cc:140] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.
[GUI] [Dbg] [Gui.cc:140] GUI requesting list of world names. The server may be busy downloading resources. Please be patient.

```

chapulina, 2021-01-13:
What Ubuntu version are you in, @ekorudiawan ?

The Qt errors looks suspicious:

`[GUI] [Wrn] [Application.cc:649] [QT] file:///usr/lib/x86_64-linux-gnu/qt5/qml/QtQuick/Dialogs/DefaultFileDialog.qml:102:33: QML Settings: Failed to initialize QSettings instance. Status code is: 1`

Failing to get the paths shouldn't affect `shapes.sdf` because that world doesn't need any custom resource paths. So I think your problem with the GUI now showing is related to those Qt errors.

ekorudiawan, 2021-01-14:
I use Linux Mint Ulyana and it is based on Ubuntu 20.04