gazebosim/gz-sim #590: Bounding boxes don't resize and follow models correctly

brawner, 2021-01-28:
Because bounding boxes are only computed when entities are selected, they don't resize and may not correctly a model as it moves during simulation.

In this video, I click the model, which is composed of two spheres without any joint between them. The bounding box is computed correctly upon clicking, but once the spheres start moving the box follows the pose of the "canonical link" sphere.

https://user-images.githubusercontent.com/1044986/106192490-1bbc9d80-6161-11eb-9673-b21c1feb081b.mp4

brawner, 2021-01-28:
This will also be an issue with nested models, since the bounding box of a nested model will follow the canonical link, which may be in a different nested model.

https://user-images.githubusercontent.com/1044986/106192958-b74e0e00-6161-11eb-936a-cb341c556272.mp4

brawner, 2021-01-28:
The behavior works as expected if instead of clicking the nested model, you click the link inside the model.

https://user-images.githubusercontent.com/1044986/106193116-ef555100-6161-11eb-9738-bd915252551e.mp4

brawner, 2021-01-28:
@EricCousineau-TRI per today's meeting

brawner, 2021-01-28:
This follows up nested model composition added in https://github.com/ignitionrobotics/ign-gazebo/pull/542

chapulina, 2021-01-28:
I remember discussing this issue back when the bounding boxes were implemented, but I can't find a record of it. I believe we decided to follow Gazebo classic's behaviour, which is what you're seeing, because recomputing the AABB on the fly would be too expensive for little gain.

brawner, 2021-01-28:
That sees reasonable, and I didn't realize this was intentional. With #542 they at least update correctly now when you click re-click the same model again.

EricCousineau-TRI, 2021-01-29:
Just for comparison - what do other tools do (stuff that has a hiearchy, has animation and/or physics, etc) - e.g. Blender?