gazebosim/gz-sim #592: Nested models which don't contain the canonical link will not have accurate poses

brawner, 2021-01-28:
In a nested model, poses are relative to the canonical link, which may be in another model. This prevents a nested model that doesn't contain the canonical link from having a correct pose and it means that the link inside that model has a pose relative to the canonical link in another model.

In this example, one sphere starts off 1 meter above an inclined plane. The canonical link is `model_00::model_10::link_00`. When simulating, the pose of `model_00::model_11::link_01` changes and is relative to `model_00::model_10::link_00`, while the pose of `model_00::model_11` remains its originally specified pose.

I would expect the pose `model_11` to update and the pose of `link_01` to remain 0 in this case.

https://user-images.githubusercontent.com/1044986/106194649-f7ae8b80-6163-11eb-8cf5-953ef0a27d17.mp4

brawner, 2021-01-28:
@EricCousineau-TRI per today's meeting

brawner, 2021-01-28:
This issue follows up https://github.com/ignitionrobotics/ign-gazebo/pull/542

EricCousineau-TRI, 2021-01-28:
I think I understand now - so the physics are as expected (both spheres's move in sensible ways), but the GUI's indication of the non-canonical-link-containing-model (and it's link) are unexpected.

So the solution would be to tell the GUI / Physics engine what to express those links relative to?

And the other question - is the time-varying position of `model_00` what you expect?

brawner, 2021-01-28:
Yes to both questions. `model_00` definitely updates as expected, since it follows the canonical link. `model_10` is also correct because the reference frame of `link_00`, `model_10` and `model_00` are all identical. The issue is with models and links that don't contain this canonical link.

azeey, 2021-07-23:
I believe this has been fixed since https://github.com/ignitionrobotics/ign-gazebo/pull/685. I've reproduce the example in the issue in case anyone else wants to test it out:

[pose_to_canonical_link.sdf.txt](https://github.com/ignitionrobotics/ign-gazebo/files/6871118/pose_to_canonical_link.sdf.txt)

closed 2021-07-23 by azeey