gazebosim/gz-sim #610: Conversions.cc error C2440 annot convert from 'const sdf::v11::Pbr *' to 'sdf::v11::Pbr *' and 'const sdf::v11::PbrWorkflow *' to 'sdf::v11::PbrWorkflow *'

sloretz, 2021-02-03:
Windows CI fails to build because of conversions from a const to a non-const type.

https://build.osrfoundation.org/job/ign_gazebo-ci-win/64/

https://github.com/ignitionrobotics/ign-gazebo/blob/d50acebae4b65df18928b0a7a9e62a252be1c608/src/Conversions.cc#L277

```
D:\Jenkins\workspace\ign_gazebo-ci-win\ws\ign-gazebo\src\Conversions.cc(277,1): error C2440: 'initializing': cannot convert from 'const sdf::v11::Pbr *' to 'sdf::v11::Pbr *' [D:\Jenkins\workspace\ign_gazebo-ci-win\ws\build\ignition-gazebo5\src\ignition-gazebo5.vcxproj]
```

https://github.com/ignitionrobotics/ign-gazebo/blob/d50acebae4b65df18928b0a7a9e62a252be1c608/src/Conversions.cc#L281

```
D:\Jenkins\workspace\ign_gazebo-ci-win\ws\ign-gazebo\src\Conversions.cc(281,1): error C2440: 'initializing': cannot convert from 'const sdf::v11::PbrWorkflow *' to 'sdf::v11::PbrWorkflow *' [D:\Jenkins\workspace\ign_gazebo-ci-win\ws\build\ignition-gazebo5\src\ignition-gazebo5.vcxproj]
```

https://github.com/ignitionrobotics/ign-gazebo/blob/d50acebae4b65df18928b0a7a9e62a252be1c608/src/Conversions.cc#L286

```

D:\Jenkins\workspace\ign_gazebo-ci-win\ws\ign-gazebo\src\Conversions.cc(286,1): error C2440: '=': cannot convert from 'const sdf::v11::PbrWorkflow *' to 'sdf::v11::PbrWorkflow *' [D:\Jenkins\workspace\ign_gazebo-ci-win\ws\build\ignition-gazebo5\src\ignition-gazebo5.vcxproj]
```

sloretz, 2021-02-03:
This issue also affects Ubuntu Bionic: https://build.osrfoundation.org/job/ignition_gazebo-ci-main-bionic-amd64/22/console

```
/var/lib/jenkins/workspace/ignition_gazebo-ci-main-bionic-amd64/ign-gazebo/src/Conversions.cc: In function ���Out ignition::gazebo::v5::convert(const sdf::v11::Material&) [with Out = ignition::msgs::Material]���:
/var/lib/jenkins/workspace/ignition_gazebo-ci-main-bionic-amd64/ign-gazebo/src/Conversions.cc:277:34: error: invalid conversion from ���const sdf::v11::Pbr*��� to ���sdf::v11::Pbr*��� [-fpermissive]
   sdf::Pbr *pbr = _in.PbrMaterial();
                   ~~~~~~~~~~~~~~~^~
/var/lib/jenkins/workspace/ignition_gazebo-ci-main-bionic-amd64/ign-gazebo/src/Conversions.cc:281:47: error: invalid conversion from ���const sdf::v11::PbrWorkflow*��� to ���sdf::v11::PbrWorkflow*��� [-fpermissive]
     sdf::PbrWorkflow *workflow = pbr->Workflow(sdf::PbrWorkflowType::METAL);
                                  ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/jenkins/workspace/ignition_gazebo-ci-main-bionic-amd64/ign-gazebo/src/Conversions.cc:286:31: error: invalid conversion from ���const sdf::v11::PbrWorkflow*��� to ���sdf::v11::PbrWorkflow*��� [-fpermissive]
       workflow = pbr->Workflow(sdf::PbrWorkflowType::SPECULAR);
```

chapulina, 2021-02-03:
Fixed by #607

closed 2021-02-12 by chapulina