gazebosim/gz-sim #619: Double free issue with when freeing unordered_maps in physics/physics.cc

brawner, 2021-02-10:
I'm finding this issue when running the `user_commands` integration test on the main branches. I haven't been able to find the exact root issue, but it appears to be a double ownership of a dartsim collision shape. This issue probably exists in other integration tests, but this one is the one I noticed it in.

The implicit destructor makes it hard to pin point the maps that are double freeing, but I believe it's the `entityCollisionMap` and one of 'entityLinkMeshMap` and `entityLinkCollisionMap`. The calls to entityCast casts the results into two separate cache maps, I think without properly incrementing shared_ptr count somewhere.

https://github.com/ignitionrobotics/ign-gazebo/blob/2bc062dd5493b3f6550e97c098c2d09ecc47b04f/src/systems/physics/Physics.cc#L922
https://github.com/ignitionrobotics/ign-gazebo/blob/2bc062dd5493b3f6550e97c098c2d09ecc47b04f/src/systems/physics/Physics.cc#L945
https://github.com/ignitionrobotics/ign-gazebo/blob/main/src/systems/physics/Physics.cc#L966

Valgrind output:
```
==1724518== 
==1724518== HEAP SUMMARY:
==1724518==     in use at exit: 316,790 bytes in 3,576 blocks
==1724518==   total heap usage: 2,054,474 allocs, 2,050,898 frees, 355,226,907 bytes allocated
==1724518== 
==1724518== LEAK SUMMARY:
==1724518==    definitely lost: 4,724 bytes in 276 blocks
==1724518==    indirectly lost: 0 bytes in 0 blocks
==1724518==      possibly lost: 2,496 bytes in 6 blocks
==1724518==    still reachable: 309,570 bytes in 3,294 blocks
==1724518==         suppressed: 0 bytes in 0 blocks
==1724518== Rerun with --leak-check=full to see details of leaked memory
==1724518== 
==1724518== ERROR SUMMARY: 18 errors from 4 contexts (suppressed: 0 from 0)
==1724518== 
==1724518== 3 errors in context 1 of 4:
==1724518== Invalid read of size 8
==1724518==    at 0x535627E2: std::_Hashtable<dart::dynamics::JacobianNode*, dart::dynamics::JacobianNode*, std::allocator<dart::dynamics::JacobianNode*>, std::__detail::_Identity, std::equal_to<dart::dynamics::JacobianNode*>, std::hash<dart::dynamics::JacobianNode*>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, true, true> >::_M_erase(std::integral_constant<bool, true>, dart::dynamics::JacobianNode* const&) (in /usr/lib/x86_64-linux-gnu/libdart.so.6.9.2)
==1724518==    by 0x535F24F0: dart::dynamics::JacobianNode::~JacobianNode() (in /usr/lib/x86_64-linux-gnu/libdart.so.6.9.2)
==1724518==    by 0x5365DC74: dart::dynamics::ShapeNode::~ShapeNode() (in /usr/lib/x86_64-linux-gnu/libdart.so.6.9.2)
==1724518==    by 0x5365DE6C: dart::dynamics::ShapeNode::~ShapeNode() (in /usr/lib/x86_64-linux-gnu/libdart.so.6.9.2)
==1724518==    by 0x5355E8D5: std::_Sp_counted_ptr<dart::dynamics::NodeDestructor*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (in /usr/lib/x86_64-linux-gnu/libdart.so.6.9.2)
==1724518==    by 0x531D1AF7: _M_release (shared_ptr_base.h:155)
==1724518==    by 0x531D1AF7: _M_release (shared_ptr_base.h:148)
==1724518==    by 0x531D1AF7: ~__shared_count (shared_ptr_base.h:730)
==1724518==    by 0x531D1AF7: ~__shared_ptr (shared_ptr_base.h:1169)
==1724518==    by 0x531D1AF7: ~shared_ptr (shared_ptr.h:103)
==1724518==    by 0x531D1AF7: ~TemplateNodePtr (NodePtr.hpp:49)
==1724518==    by 0x531D1AF7: ~ShapeInfo (Base.hh:71)
==1724518==    by 0x531D1AF7: destroy<ignition::physics::dartsim::ShapeInfo> (new_allocator.h:153)
==1724518==    by 0x531D1AF7: destroy<ignition::physics::dartsim::ShapeInfo> (alloc_traits.h:497)
==1724518==    by 0x531D1AF7: std::_Sp_counted_ptr_inplace<ignition::physics::dartsim::ShapeInfo, std::allocator<ignition::physics::dartsim::ShapeInfo>, (__gnu_cxx::_Lock_policy)2>::_M_dispose() (shared_ptr_base.h:557)
==1724518==    by 0x14D927: _M_release (shared_ptr_base.h:155)
==1724518==    by 0x14D927: std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() (shared_ptr_base.h:148)
==1724518==    by 0x5199E383: ~__shared_count (shared_ptr_base.h:730)
==1724518==    by 0x5199E383: ~__shared_ptr (shared_ptr_base.h:1169)
==1724518==    by 0x5199E383: ~shared_ptr (shared_ptr.h:103)
==1724518==    by 0x5199E383: ~Identity (Identity.hh:73)
==1724518==    by 0x5199E383: ~Entity (Entity.hh:294)
==1724518==    by 0x5199E383: ~Shape (FeatureList.hh:698)
==1724518==    by 0x5199E383: _M_destroy (optional:257)
==1724518==    by 0x5199E383: _M_reset (optional:277)
==1724518==    by 0x5199E383: ~_Optional_payload (optional:398)
==1724518==    by 0x5199E383: ~_Optional_base (optional:471)
==1724518==    by 0x5199E383: ~optional (optional:656)
==1724518==    by 0x5199E383: ~EntityPtr (Entity.hh:47)
==1724518==    by 0x5199E383: ~pair (stl_pair.h:208)
==1724518==    by 0x5199E383: destroy<std::pair<long unsigned int const, ignition::physics::EntityPtr<ignition::physics::Shape<ignition::physics::FeaturePolicy<double, 3>, ignition::physics::FeatureList<ignition::physics::FeatureList<ignition::gazebo::v5::systems::PhysicsPrivate::MinimumFeatureList, ignition::physics::GetContactsFromLastStepFeature, ignition::physics::sdf::ConstructSdfCollision>, ignition::physics::CollisionFilterMaskFeature> > > > > (new_allocator.h:153)
==1724518==    by 0x5199E383: destroy<std::pair<long unsigned int const, ignition::physics::EntityPtr<ignition::physics::Shape<ignition::physics::FeaturePolicy<double, 3>, ignition::physics::FeatureList<ignition::physics::FeatureList<ignition::gazebo::v5::systems::PhysicsPrivate::MinimumFeatureList, ignition::physics::GetContactsFromLastStepFeature, ignition::physics::sdf::ConstructSdfCollision>, ignition::physics::CollisionFilterMaskFeature> > > > > (alloc_traits.h:497)
==1724518==    by 0x5199E383: std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<unsigned long const, ignition::physics::EntityPtr<ignition::physics::Shape<ignition::physics::FeaturePolicy<double, 3ul>, ignition::physics::FeatureList<ignition::physics::FeatureList<ignition::gazebo::v5::systems::PhysicsPrivate::MinimumFeatureList, ignition::physics::GetContactsFromLastStepFeature, ignition::physics::sdf::ConstructSdfCollision>, ignition::physics::CollisionFilterMaskFeature> > > >, false> > >::_M_deallocate_node(std::__detail::_Hash_node<std::pair<unsigned long const, ignition::physics::EntityPtr<ignition::physics::Shape<ignition::physics::FeaturePolicy<double, 3ul>, ignition::physics::FeatureList<ignition::physics::FeatureList<ignition::gazebo::v5::systems::PhysicsPrivate::MinimumFeatureList, ignition::physics::GetContactsFromLastStepFeature, ignition::physics::sdf::ConstructSdfCollision>, ignition::physics::CollisionFilterMaskFeature> > > >, false>*) [clone .isra.0] (hashtable_policy.h:2102)
==1724518==    by 0x519CD6DA: _M_deallocate_nodes (hashtable_policy.h:2124)
==1724518==    by 0x519CD6DA: clear (hashtable.h:2028)
==1724518==    by 0x519CD6DA: ~_Hashtable (hashtable.h:1352)
==1724518==    by 0x519CD6DA: ~unordered_map (unordered_map.h:102)
==1724518==    by 0x519CD6DA: ignition::gazebo::v5::systems::PhysicsPrivate::~PhysicsPrivate() (Physics.cc:107)
==1724518==    by 0x5199E9D3: operator() (unique_ptr.h:81)
==1724518==    by 0x5199E9D3: ~unique_ptr (unique_ptr.h:292)
==1724518==    by 0x5199E9D3: ignition::gazebo::v5::systems::Physics::~Physics() (Physics.hh:64)
==1724518==    by 0x5199E9FC: ignition::gazebo::v5::systems::Physics::~Physics() (Physics.hh:64)
==1724518==    by 0x4BA7F95: operator() (std_function.h:688)
[... 408 lines cut ...]
==1724518==    by 0x4BA6BDF: operator() (unique_ptr.h:81)
==1724518==    by 0x4BA6BDF: ~unique_ptr (unique_ptr.h:292)
==1724518==    by 0x4BA6BDF: ignition::plugin::Plugin::~Plugin() (Plugin.cc:385)
==1724518==  Block was alloc'd at
==1724518==    at 0x483B7F3: malloc (vg_replace_malloc.c:309)
==1724518==    by 0x531C9327: aligned_malloc (Memory.h:159)
==1724518==    by 0x531C9327: conditional_aligned_malloc<true> (Memory.h:214)
==1724518==    by 0x531C9327: operator new (BodyNode.hpp:1199)
==1724518==    by 0x531C9327: createJointAndBodyNodePair<dart::dynamics::FreeJoint> (Skeleton.hpp:83)
==1724518==    by 0x531C9327: ignition::physics::dartsim::SDFFeatures::ConstructSdfLink(ignition::physics::Identity const&, sdf::v11::Link const&) (SDFFeatures.cc:459)
==1724518==    by 0x519D2BE3: ignition::physics::sdf::ConstructSdfLink::Model<ignition::physics::FeaturePolicy<double, 3ul>, ignition::gazebo::v5::systems::PhysicsPrivate::MinimumFeatureList>::ConstructLink(sdf::v11::Link const&) (ConstructLink.hh:54)
[... cut ...]
==1724518==    by 0x4A33E1A: ignition::gazebo::v5::SimulationRunner::Step(ignition::gazebo::v5::UpdateInfo const&) (SimulationRunner.cc:728)
==1724518==    by 0x4A3468A: ignition::gazebo::v5::SimulationRunner::Run(unsigned long) (SimulationRunner.cc:693)
==1724518==    by 0x4A281EB: ignition::gazebo::v5::ServerPrivate::Run(unsigned long, std::optional<std::condition_variable*>) (ServerPrivate.cc:165)
==1724518==    by 0x4A1F2AB: ignition::gazebo::v5::Server::Run(bool, unsigned long, bool) (Server.cc:235)
==1724518== 
==1724518== ERROR SUMMARY: 18 errors from 4 contexts (suppressed: 0 from 0)

```

chapulina, 2021-02-10:
This may be fixed by #586

brawner, 2021-02-10:
Ooh, I think so. Sounds like this issue in dart is also a part of the issue. https://github.com/dartsim/dart/issues/1534

chapulina, 2021-11-15:
I suspect this has been fixed, please reopen if I'm mistaken

closed 2021-11-15 by chapulina