gazebosim/gz-sim #630: ign-gazebo3 homebrew INTEGRATION test failures

jacobperron, 2021-02-12:
## Environment
* OS Version: macOS
* Source or binary build?
  3465bbe

## Description

Three new test failures over the last couple days: https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/113/testReport/

The follow two started today:

<details>
<summary>INTEGRATION_each_new_removed (fails with exception)</summary>
<pre>
        Start  67: INTEGRATION_each_new_removed

67: Test command: /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/build/bin/INTEGRATION_each_new_removed "--gtest_output=xml:/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/build/test_results/INTEGRATION_each_new_removed.xml"
67: Test timeout computed to be: 240
67: Running main() from /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/ign-gazebo/test/gtest/src/gtest_main.cc
67: [==========] Running 1 test from 1 test suite.
67: [----------] Global test environment set-up.
67: [----------] 1 test from EachNewRemovedFixture
67: [ RUN      ] EachNewRemovedFixture.EachNewEachRemovedInSystem
 67/150 Test  #67: INTEGRATION_each_new_removed .........................Subprocess aborted***Exception:   1.08 sec
</pre>
</details>

<details>
<summary>INTEGRATION_performer_detector (fails with segfault)</summary>
<pre>
103: [ RUN      ] PerformerDetectorTest.HandlesRemovedParentModel
103: [Err] [PerformerDetector.cc:104] Performer detector[detector1] has an empty <value> with an associated <key> of [no_value]. This <header_data> will be ignored.
103: [Err] [PerformerDetector.cc:98] Performer detector[detector1] has an empty <key> with an associated <value> of [no_key]. This <header_data> will be ignored.
103/150 Test #103: INTEGRATION_performer_detector .......................***Exception: SegFault  1.85 sec
</pre>
</details>

There are a two recent changes to the `ign-gazebo3` branch, though I'm not sure if either of them is responsible for the recent failures:

- https://github.com/ignitionrobotics/ign-gazebo/pull/603
- https://github.com/ignitionrobotics/ign-gazebo/pull/574

This third failure started two days ago:

<details>
<summary>INTEGRATION_breadcrumbs (fails with segfault)</summary>
<pre>
55: [ RUN      ] BreadcrumbsTest.FuelDeploy
55: [Msg] Loading SDF world file[/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/ign-gazebo/test/worlds/breadcrumbs.sdf].
55: Warning [Model.cc:212] Non-unique names detected in XML children of model with name[vehicle_blue].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [17459188283658606303] / [ign_gazebo_components.RenderEngineServerPlugin].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [8705992680619689917] / [ign_gazebo_components.RenderEngineGuiPlugin].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [17605309075052480649] / [ign_gazebo_components.Atmosphere].
55: [Dbg] [LevelManager.cc:195] Reading performer info
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [4425034272867445569] / [ign_gazebo_components.Performer].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [11311330372155309806] / [ign_gazebo_components.PerformerLevels].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [17121648710877364109] / [ign_gazebo_components.Geometry].
55: [Msg] Created performer [3 / perf1]
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [8064491505919932473] / [ign_gazebo_components.Level].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [2668898242563798256] / [ign_gazebo_components.DefaultLevel].
[... 58 lines cut ...]
55: [Msg] Found no publishers on /clock, adding root clock topic
55: [Dbg] [SimulationRunner.cc:422] Creating PostUpdate worker threads: 4
55: [Dbg] [SimulationRunner.cc:435] Creating postupdate worker thread (0)
55: [Dbg] [SimulationRunner.cc:435] Creating postupdate worker thread (1)
55: [Dbg] [SimulationRunner.cc:435] Creating postupdate worker thread (2)
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [14423792621074650228] / [ign_gazebo_components.JointVelocityCmd].
55: [Dbg] [EntityComponentManager.cc:632] Using components of type [9436740692853731264] / [ign_gazebo_components.JointPosition].
55: [Msg] Serving scene information on [/world/breadcrumbs/scene/info]
55: [Msg] Serving graph information on [/world/breadcrumbs/scene/graph]
55: [Msg] Serving full state on [/world/breadcrumbs/state]
55: [Msg] Serving full state (async) on [/world/breadcrumbs/state_async]
55: [Msg] Publishing scene information on [/world/breadcrumbs/scene/info]
55: [Msg] Publishing entity deletions on [/world/breadcrumbs/scene/deletion]
55: [Msg] Publishing state changes on [/world/breadcrumbs/state]
55: [Msg] Publishing pose messages on [/world/breadcrumbs/pose/info]
55: [Msg] Publishing dynamic pose messages on [/world/breadcrumbs/dynamic_pose/info]
55: [Msg] Deploying B2_0 at -2.2 2 5.525 -0 0 0
55: [Msg] Deploying B2_1 at -2.2 2 5.525 -0 0 0
55: [Msg] Deploying B2_2 at -2.2 2 5.525 -0 0 0
55: [Msg] Deploying B2_3 at -2.2 2 5.525 -0 0 -0[1
 55/150 Test  #55: INTEGRATION_breadcrumbs ..............................***Exception: SegFault  4.58 sec
</pre>
</details>

Correlates to https://github.com/ignitionrobotics/ign-gazebo/pull/609, but I'm just speculating that they may be related.

j-rivero, 2021-04-08:
I've detected [some more today](https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo4-homebrew-amd64/59):

```
The following tests FAILED:
	 61 - INTEGRATION_breadcrumbs (SEGFAULT)
	 75 - INTEGRATION_each_new_removed (SEGFAULT)
	143 - INTEGRATION_user_commands (SEGFAULT)
```

j-rivero, 2021-07-07:
Current list in `main` branch is:
```

	 61 - INTEGRATION_breadcrumbs (SEGFAULT)
	151 - INTEGRATION_user_commands (SEGFAULT)
```

chapulina, 2021-07-07:
https://github.com/ignitionrobotics/ign-gazebo/pull/807 has a potential solution

jacobperron, 2021-09-15:
More segfaulting tests from yesterday

```
	 59 - INTEGRATION_ackermann_steering_system (SEGFAULT)
	119 - INTEGRATION_logical_audio_sensor_plugin (SEGFAULT)
	159 - INTEGRATION_user_commands (SEGFAULT)
```

https://build.osrfoundation.org/job/ignition_gazebo-ci-main-homebrew-amd64/79

Blast545, 2022-06-20:
Grepping our last 30 days jobs at https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/ we find the following Integration tests segfaulting:

```
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|221|INTEGRATION_pose_publisher_system
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|219|INTEGRATION_diff_drive_system
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|219|INTEGRATION_velocity_control_system
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|212|INTEGRATION_pose_publisher_system
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|212|INTEGRATION_touch_plugin
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|211|INTEGRATION_altimeter_system
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|209|INTEGRATION_level_manager
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|208|INTEGRATION_diff_drive_system
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|208|INTEGRATION_level_manager
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|207|INTEGRATION_network_handshake
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|206|INTEGRATION_pose_publisher_system
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|205|INTEGRATION_logical_audio_sensor_plugin
ignition_gazebo-ci-ign-gazebo3-homebrew-amd64|205|INTEGRATION_performer_detector
```

I'm opening a PR to disable them on MacOS for now.

azeey, 2024-08-08:
I think most of these segfaults have been fixed by https://github.com/gazebosim/gz-sim/pull/1836. I'll go ahead and close this.

closed 2024-08-08 by azeey