gazebosim/gz-sim #634: User command light test failure

chapulina, 2021-02-17:
## Environment
* OS Version: all
* Source or binary build? CI builds from source

## Description

This was introduced on https://github.com/ignitionrobotics/ign-gazebo/pull/482, I was a bit too quick to press the merge button.

## Steps to reproduce

Run `INTEGRATION_user_commands` test.

## Output

Test failure message:

<details>
  <summary>UserCommandsTest.Light</summary>
  
```
   [ RUN      ] UserCommandsTest.Light
  [Msg] Loading SDF world file[/github/workspace/test/worlds/lights_render.sdf].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [2188341333082264598] / [ign_gazebo_components.Physics].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [17459188283658606303] / [ign_gazebo_components.RenderEngineServerPlugin].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [8705992680619689917] / [ign_gazebo_components.RenderEngineGuiPlugin].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [17605309075052480649] / [ign_gazebo_components.Atmosphere].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [8064491505919932473] / [ign_gazebo_components.Level].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [2668898242563798256] / [ign_gazebo_components.DefaultLevel].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [11371360182141354106] / [ign_gazebo_components.LevelEntityNames].
  [Dbg] [SimulationRunner.cc:839] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
  [Msg] Create service on [/world/lights_command/create]
  [Msg] Remove service on [/world/lights_command/remove]
  [Msg] Pose service on [/world/lights_command/set_pose]
  [Msg] Light configuration service on [/world/lights_command/light_config]
  [Msg] Physics service on [/world/lights_command/set_physics]
  [Dbg] [SimulationRunner.cc:839] Loaded system [ignition::gazebo::systems::UserCommands] for entity [1]
  Error while loading the library [/usr/local/lib/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so]: libignition-gazebo5-rendering.so.5: cannot open shared object file: No such file or directory
  [Err] [SystemLoader.cc:75] Failed to load system plugin [ignition-gazebo-sensors-system] : couldn't load library on path [/usr/local/lib/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [4981278897826323946] / [ign_gazebo_components.WorldSdf].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [6687176221774458630] / [ign_gazebo_components.Model].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [6612894081701502240] / [ign_gazebo_components.Pose].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [8546580419506082455] / [ign_gazebo_components.Static].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [9712747055438129860] / [ign_gazebo_components.WindMode].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [5661073481138181711] / [ign_gazebo_components.SelfCollide].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [11683062252779233161] / [ign_gazebo_components.SourceFilePath].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [5081358965268446661] / [ign_gazebo_components.Link].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [8112400427272910195] / [ign_gazebo_components.Inertial].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [16454635107327670381] / [ign_gazebo_components.Visual].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [13011964647677164955] / [ign_gazebo_components.CastShadows].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [13440282432131634483] / [ign_gazebo_components.Transparency].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [5453622280849253520] / [ign_gazebo_components.VisibilityFlags].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [17121648710877364109] / [ign_gazebo_components.Geometry].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [3866641186784191835] / [ign_gazebo_components.Light].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [273039548655896271] / [ign_gazebo_components.Sensor].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [14624144930658477562] / [ign_gazebo_components.Camera].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [10522242218202596205] / [ign_gazebo_components.CanonicalLink].
  [Dbg] [EntityComponentManager.cc:672] Using components of type [11536476718181283925] / [ign_gazebo_components.ModelSdf].
  [Msg] Loaded level [3]
  [Msg] Serving world controls on [/world/lights_command/control] and [/world/lights_command/playback/control]
[... 234 lines cut ...]
  [Msg] Serving full state on [/world/empty/state]
  [Msg] Serving full state (async) on [/world/empty/state_async]
  [Msg] Publishing scene information on [/world/empty/scene/info]
  [Msg] Publishing entity deletions on [/world/empty/scene/deletion]
  [Msg] Publishing state changes on [/world/empty/state]
  [Msg] Publishing pose messages on [/world/empty/pose/info]
  [Msg] Publishing dynamic pose messages on [/world/empty/dynamic_pose/info]
  [Dbg] [UserCommands.cc:721] Created entity [9] named [spawned_model]
  [Dbg] [UserCommands.cc:721] Created entity [13] named [spawned_model_0]
  [Dbg] [UserCommands.cc:721] Created entity [17] named [banana]
  [Dbg] [UserCommands.cc:721] Created entity [21] named [spawned_light]
  [Dbg] [UserCommands.cc:721] Created entity [22] named [light_test]
  [Dbg] [UserCommands.cc:721] Created entity [23] named [acerola]
  [Dbg] [UserCommands.cc:721] Created entity [27] named [coconut]
  [Dbg] [UserCommands.cc:721] Created entity [31] named [accepted_light]
  Error [parser.cc:562] Error parsing XML from string: Error=XML_ERROR_MISMATCHED_ELEMENT ErrorID=16 (0x10) Line number=1: XMLElement name=sdf
  [Err] [UserCommands.cc:598] Error Code 28 Msg: Unable to read SDF string: <?xml version='1.0' ?><sdf version='1.6'></sdfo>
  [Dbg] [UserCommands.cc:721] Created entity [32] named [test_model]
  [Dbg] [SimulationRunner.cc:499] Exiting postupdate worker thread (1)
  [Dbg] [SimulationRunner.cc:499] Exiting postupdate worker thread (2)
  [Dbg] [SimulationRunner.cc:499] Exiting postupdate worker thread (0)
  [       OK ] UserCommandsTest.Create (1609 ms)
```  

</details>

ahcorde, 2021-02-18:
Backtrace

<details>
  <summary>Backtrace</summary>

```
0x00007fffd63247f4 in std::_Hashtable<dart::dynamics::JacobianNode*, dart::dynamics::JacobianNode*, std::allocator<dart::dynamics::JacobianNode*>, std::__detail::_Identity, std::equal_to<dart::dynamics::JacobianNode*>, std::hash<dart::dynamics::JacobianNode*>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, true, true> >::_M_erase(std::integral_constant<bool, true>, dart::dynamics::JacobianNode* const&) () from /usr/lib/x86_64-linux-gnu/libdart.so.6
(gdb) bt
#0  0x00007fffd63247f4 in std::_Hashtable<dart::dynamics::JacobianNode*, dart::dynamics::JacobianNode*, std::allocator<dart::dynamics::JacobianNode*>, std::__detail::_Identity, std::equal_to<dart::dynamics::JacobianNode*>, std::hash<dart::dynamics::JacobianNode*>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, true, true> >::_M_erase(std::integral_constant<bool, true>, dart::dynamics::JacobianNode* const&) () from /usr/lib/x86_64-linux-gnu/libdart.so.6
#1  0x00007fffd63b44f1 in dart::dynamics::JacobianNode::~JacobianNode() () from /usr/lib/x86_64-linux-gnu/libdart.so.6
#2  0x00007fffd641fc75 in dart::dynamics::ShapeNode::~ShapeNode() () from /usr/lib/x86_64-linux-gnu/libdart.so.6
#3  0x00007fffd641fe6d in dart::dynamics::ShapeNode::~ShapeNode() () from /usr/lib/x86_64-linux-gnu/libdart.so.6
#4  0x00007fffd63208d6 in std::_Sp_counted_ptr<dart::dynamics::NodeDestructor*, (__gnu_cxx::_Lock_policy)2>::_M_dispose() () from /usr/lib/x86_64-linux-gnu/libdart.so.6
#5  0x00007fffd675f9d8 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5555570d6860) at /usr/include/c++/9/bits/shared_ptr_base.h:148
#6  std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5555570d6860) at /usr/include/c++/9/bits/shared_ptr_base.h:148
#7  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:730
#8  std::__shared_ptr<dart::dynamics::NodeDestructor, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#9  std::shared_ptr<dart::dynamics::NodeDestructor>::~shared_ptr (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr.h:103
#10 dart::dynamics::TemplateNodePtr<dart::dynamics::ShapeNode, dart::dynamics::BodyNode>::~TemplateNodePtr (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/dart/dynamics/detail/NodePtr.hpp:49
#11 ignition::physics::dartsim::ShapeInfo::~ShapeInfo (this=<optimized out>, __in_chrg=<optimized out>) at /home/ahcorde/ignition_dome/src/ign-physics/dartsim/src/Base.hh:71
#12 __gnu_cxx::new_allocator<ignition::physics::dartsim::ShapeInfo>::destroy<ignition::physics::dartsim::ShapeInfo> (this=<optimized out>, __p=<optimized out>) at /usr/include/c++/9/ext/new_allocator.h:153
#13 std::allocator_traits<std::allocator<ignition::physics::dartsim::ShapeInfo> >::destroy<ignition::physics::dartsim::ShapeInfo> (__a=..., __p=<optimized out>) at /usr/include/c++/9/bits/alloc_traits.h:497
#14 std::_Sp_counted_ptr_inplace<ignition::physics::dartsim::ShapeInfo, std::allocator<ignition::physics::dartsim::ShapeInfo>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:557
#15 0x00005555555a05e8 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5555570f1710) at /usr/include/c++/9/bits/shared_ptr_base.h:148
#16 std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x5555570f1710) at /usr/include/c++/9/bits/shared_ptr_base.h:148
#17 0x00007fffec6f65a4 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x5555570f4630, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#18 std::__shared_ptr<void, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x5555570f4628, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#19 std::shared_ptr<void>::~shared_ptr (this=0x5555570f4628, __in_chrg=<optimized out>) at /usr/include/c++/9/bits/shared_ptr.h:103
#20 ignition::physics::Identity::~Identity (this=0x5555570f4620, __in_chrg=<optimized out>) at /home/ahcorde/ignition_dome/install/include/ignition/physics3/ignition/physics/detail/Identity.hh:73
#21 ignition::physics::Entity<ignition::physics::FeaturePolicy<double, 3ul>, ignition::physics::FeatureList<ignition::physics::FeatureList<ignition::gazebo::v4::systems::PhysicsPrivate::MinimumFeatureList, ignition::physics::GetContactsFromLastStepFeature, ignition::physics::sdf::ConstructSdfCollision>, ignition::physics::CollisionFilterMaskFeature> >::~Entity (this=0x5555570f4608, __in_chrg=<optimized out>)
    at /home/ahcorde/ignition_dome/install/include/ignition/physics3/ignition/physics/Entity.hh:294
#22 ignition::physics::Shape<ignition::physics::FeaturePolicy<double, 3ul>, ignition::physics::FeatureList<ignition::physics::FeatureList<ignition::gazebo::v4::systems::PhysicsPrivate::MinimumFeatureList, ignition::physics::GetContactsFromLastStepFeature, ignition::physics::sdf::ConstructSdfCollision>, ignition::physics::CollisionFilterMaskFeature> >::~Shape (this=0x5555570f4600, __in_chrg=<optimized out>, 
[... cut ...]
    location=location@entry=0x555555607158 "auxiliary test code (environments or event listeners)") at /home/ahcorde/ignition_dome/src/ign-gazebo/test/gtest/src/gtest.cc:2473
#88 0x00005555555f2935 in testing::UnitTest::Run (this=0x555555629d20 <testing::UnitTest::GetInstance()::instance>) at /home/ahcorde/ignition_dome/src/ign-gazebo/test/gtest/include/gtest/gtest.h:1412
#89 0x000055555558ce54 in RUN_ALL_TESTS () at /home/ahcorde/ignition_dome/src/ign-gazebo/test/gtest/include/gtest/gtest.h:2473
#90 main (argc=<optimized out>, argv=0x7fffffffdc18) at /home/ahcorde/ignition_dome/src/ign-gazebo/test/gtest/src/gtest_main.cc:45
```

</details>

ahcorde, 2021-02-18:
I simplified the backtrace
```
Server::~Server
ServerPrivate::~ServerPrivate
SimulationRunner::~SimulationRunner
SystemInternal::~SystemInternal
Plugin::~Plugin
PluginWithDlHandle::~PluginWithDlHandle 
gazebo::v4::systems::Physics::~Physics
gazebo::v4::systems::PhysicsPrivate::~PhysicsPrivate
ignition::physics::Identity::~Identity
ignition::physics::dartsim::ShapeInfo::~ShapeInfo
dart::dynamics::ShapeNode::~ShapeNode()
```

I have forced Ignition-Gazebo to use `tpe` instead of `dartsim` in this test. it seems that ign-physics is generating this problem, Destruction of the objects is wrong, I will digg in `ign-physics`. Most of these destructors are the default ones.

@scpeters any idea or thoughts ?

ahcorde, 2021-02-18:
I runned the ign-physics3 with 

<details>
  <summary>INTEGRATION_DoublePendulum</summary>

```
17: Test command: /home/ahcorde/ignition_dome/build/ignition-physics3/bin/INTEGRATION_DoublePendulum "--gtest_output=xml:/home/ahcorde/ignition_dome/build/ignition-physics3/test_results/INTEGRATION_DoublePendulum.xml"
17: Test timeout computed to be: 240
17: [==========] Running 1 test from 1 test case.
17: [----------] Global test environment set-up.
17: [----------] 1 test from DoublePendulum
17: [ RUN      ] DoublePendulum.Step
17: DoublePendulum: testing plugin: mock::dart::DARTDoublePendulum
17: Warning: link 'link1' material 'black' undefined.
17:          at line 84 in /build/urdfdom-VnCcob/urdfdom-1.0.4+ds/urdf_parser/src/model.cpp
17: Warning: link 'link1' material 'black' undefined.
17:          at line 84 in /build/urdfdom-VnCcob/urdfdom-1.0.4+ds/urdf_parser/src/model.cpp
17: Warning: link 'link2' material 'orange' undefined.
17:          at line 84 in /build/urdfdom-VnCcob/urdfdom-1.0.4+ds/urdf_parser/src/model.cpp
17: Warning: link 'link2' material 'orange' undefined.
17:          at line 84 in /build/urdfdom-VnCcob/urdfdom-1.0.4+ds/urdf_parser/src/model.cpp
17: =================================================================
17: ==280118==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60200040ef08 at pc 0x7ffaa3f89a55 bp 0x7ffdf9fe6940 sp 0x7ffdf9fe6930
17: READ of size 8 at 0x60200040ef08 thread T0
17:     #0 0x7ffaa3f89a54 in Eigen::internal::handmade_aligned_free(void*) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:98
17:     #1 0x7ffaa3f89a54 in Eigen::internal::aligned_free(void*) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:179
17:     #2 0x7ffaa3f89a54 in void Eigen::internal::conditional_aligned_free<true>(void*) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:230
17:     #3 0x7ffaa3f89a54 in void Eigen::internal::conditional_aligned_delete_auto<double, true>(double*, unsigned long) /usr/include/eigen3/Eigen/src/Core/util/Memory.h:416
17:     #4 0x7ffaa3f89a54 in Eigen::DenseStorage<double, -1, -1, 1, 0>::~DenseStorage() /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:542
17:     #5 0x7ffaa3f89a54 in Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::~PlainObjectBase() /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:98
17:     #6 0x7ffaa3f89a54 in Eigen::Matrix<double, -1, 1, 0, -1, 1>::~Matrix() /usr/include/eigen3/Eigen/src/Core/Matrix.h:178
17:     #7 0x7ffaa3f89a54 in dart::dynamics::Skeleton::Configuration::~Configuration() /usr/include/dart/dynamics/Skeleton.hpp:94
17:     #8 0x7ffaa3f89a54 in mock::dart::DARTDoublePendulum::Write(ignition::physics::JointPositions&) const /home/ahcorde/ignition_dome/src/ign-physics/test/plugins/DARTDoublePendulum.cc:276
17:     #9 0x7ffaa3f89a54 in mock::dart::DARTDoublePendulum::WorldForwardStep(ignition::physics::Identity const&, ignition::physics::SpecifyData<ignition::physics::RequireData<ignition::physics::WorldPoses>, ignition::physics::ExpectData<ignition::physics::Contacts, ignition::physics::JointPositions> >&, ignition::physics::CompositeData&, ignition::physics::ExpectData<ignition::physics::ApplyExternalForceTorques, ignition::physics::ApplyGeneralizedForces, ignition::physics::VelocityControlCommands, ignition::physics::ServoControlCommands> const&) /home/ahcorde/ignition_dome/src/ign-physics/test/plugins/DARTDoublePendulum.cc:268
17:     #10 0x55ae4de10a58 in ignition::physics::ForwardStep::World<ignition::physics::FeaturePolicy<double, 3ul>, ignition::physics::FeatureList<ignition::physics::ForwardStep, ignition::physics::GetEngineInfo, ignition::physics::GetWorldFromEngine> >::Step(ignition::physics::SpecifyData<ignition::physics::RequireData<ignition::physics::WorldPoses>, ignition::physics::ExpectData<ignition::physics::Contacts, ignition::physics::JointPositions> >&, ignition::physics::CompositeData&, ignition::physics::ExpectData<ignition::physics::ApplyExternalForceTorques, ignition::physics::ApplyGeneralizedForces, ignition::physics::VelocityControlCommands, ignition::physics::ServoControlCommands> const&) /home/ahcorde/ignition_dome/src/ign-physics/include/ignition/physics/ForwardStep.hh:162
17:     #11 0x55ae4de10a58 in DoublePendulum_TEST(ignition::plugin::TemplatePluginPtr<ignition::plugin::Plugin>) /home/ahcorde/ignition_dome/src/ign-physics/test/integration/DoublePendulum.cc:79
17:     #12 0x55ae4de1752e in DoublePendulum_Step_Test::TestBody() /home/ahcorde/ignition_dome/src/ign-physics/test/integration/DoublePendulum.cc:49
17:     #13 0x55ae4df22af9 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/ahcorde/ignition_dome/src/ign-physics/test/gtest/src/gtest.cc:2421
17:     #14 0x55ae4df22af9 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/ahcorde/ignition_dome/src/ign-physics/test/gtest/src/gtest.cc:2457
17:     #15 0x55ae4def6379 in testing::Test::Run() /home/ahcorde/ignition_dome/src/ign-physics/test/gtest/src/gtest.cc:2495
17:     #16 0x55ae4def6a39 in testing::Test::Run() /home/ahcorde/ignition_dome/src/ign-physics/test/gtest/src/gtest.cc:2487
17:     #17 0x55ae4def6a39 in testing::TestInfo::Run() /home/ahcorde/ignition_dome/src/ign-physics/test/gtest/src/gtest.cc:2671
[... cut ...]
60:   Stack after return:      f5
60:   Stack use after scope:   f8
60:   Global redzone:          f9
60:   Global init order:       f6
60:   Poisoned by user:        f7
60:   Container overflow:      fc
60:   Array cookie:            ac
60:   Intra object redzone:    bb
60:   ASan internal:           fe
60:   Left alloca redzone:     ca
60:   Right alloca redzone:    cb
60:   Shadow gap:              cc
60: ==280250==ABORTING
60/87 Test #60: UNIT_SimulationFeatures_TEST ................***Failed    0.44 sec
```
</details>

azeey, 2021-02-22:
I think this is related https://github.com/ignitionrobotics/ign-gazebo/pull/586

chapulina, 2021-02-25:
I can't reproduce the failures locally on Ubuntu.

@ahcorde , how did you get that backtrace? It seems to be crashing on the `Remove` test for you, but that's not what is happening on CI.

On macOS on Jenkins, the `Light` test is segfaulting.

On GitHub actions, the `Light` test has lots of failing expectations, and I believe it's because of this:

```
Error while loading the library [/usr/local/lib/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so]: libignition-gazebo5-rendering.so.5: cannot open shared object file: No such file or directory
  [Err] [SystemLoader.cc:75] Failed to load system plugin [ignition-gazebo-sensors-system] : couldn't load library on path [/usr/local/lib/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so].
```

See #650 for a fix - it works on GitHub actions, I haven't checked macOS yet.

chapulina, 2021-03-01:
macOS isn't fixed yet

chapulina, 2021-11-15:
Disabling the test on macOS: https://github.com/ignitionrobotics/ign-gazebo/pull/1204

azeey, 2021-11-15:
Just thought it might be related, but I've been able to get consistent segfaults when running the following if I build `ign-gazebo` using clang. 

```
bin/INTEGRATION_user_commands --gtest_filter=*Light:*Pose
```

The failure appears to be related to the unloading of a shared library from a previous test affecting the component storage map `storagesById`: https://github.com/ignitionrobotics/ign-gazebo/blob/f8942cd9b19cf0b61a1d5a7fb9f47d0c1d558e84/include/ignition/gazebo/components/Factory.hh#L310.

azeey, 2021-11-16:
> Just thought it might be related, but I've been able to get consistent segfaults when running the following if I build `ign-gazebo` using clang.
>
> ```
> bin/INTEGRATION_user_commands [...]

I should have said using clang on Linux.

PR #650 (2021-02-25): 👩‍🌾 Set LD_LIBRARY_PATH on Actions CI (fix user_commands test)
# 🦟 Bug fix

Fixes #634 on GitHub actions (possibly on Homebrew too?)

## Summary

The test failures on GitHub actions print:

```
Error while loading the library [/usr/local/lib/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so]: libignition-gazebo5-rendering.so.5: cannot open shared object file: No such file or directory
  [Err] [SystemLoader.cc:75] Failed to load system plugin [ignition-gazebo-sensors-system] : couldn't load library on path [/usr/local/lib/ign-gazebo-5/plugins/libignition-gazebo-sensors-system.so].
```

 [... cut]