gazebosim/gz-sim #638: Toggling light visualization in ign-gazebo5

ahcorde, 2021-02-18:
## Desired behavior

When the Light editor is forward ported to ign-gazebo5 we should allow to toggle light visualization in ign-gazebo5

## Additional context

Refer to these PRs https://github.com/ignitionrobotics/ign-gazebo/pull/482 and https://github.com/ignitionrobotics/ign-rendering/pull/202 for more context

ahcorde, 2022-02-25:
Related PRs
 - https://github.com/ignitionrobotics/sdformat/pull/851
 - https://github.com/ignitionrobotics/ign-gazebo/pull/1343

chapulina, 2022-03-10:
Thanks for the on/off feature, @ahcorde !

For the visualization itself, I think it would be interesting to have a `<visualize>` SDF parameter like we have for [sensors](http://sdformat.org/spec?ver=1.9&elem=sensor#sensor_visualize), then toggle that from the GUI.

ahcorde, 2022-03-11:
Allow to enable/disable light visuals

 - https://github.com/ignitionrobotics/sdformat/pull/877
 - https://github.com/ignitionrobotics/ign-gazebo/pull/1387

closed 2022-03-18 by chapulina