gazebosim/gz-sim #647: Ignition Gazebo crashes when running the example actor.sdf world

adlarkin, 2021-02-24:
## Environment
* OS Version: Ubuntu 18.04.5
* Source or binary build? Binary, Ignition Dome (Ignition Gazebo 4.5.0)

## Description
* Expected behavior: The [actor.sdf world](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/examples/worlds/actor.sdf) should load
* Actual behavior: Gazebo is never started due to a crash

## Steps to reproduce

1. run `ign gazebo actor.sdf` or `ign gazebo -r actor.sdf`

## Output
A very long backtrace is given:
```
/usr/lib/ruby/ignition/cmdgazebo4.rb:431: [BUG] Segmentation fault at 0x0000000000000100
ruby 2.5.1p57 (2018-03-29 revision 63029) [x86_64-linux-gnu]

-- Control frame information -----------------------------------------------
c:0007 p:---- s:0075 e:000074 CFUNC  :call
c:0006 p:0017 s:0053 e:000052 METHOD /usr/lib/ruby/ignition/cmdgazebo4.rb:431
c:0005 p:0160 s:0047 e:000046 BLOCK  /usr/lib/ruby/ignition/cmdgazebo4.rb:460 [FINISH]
c:0004 p:---- s:0044 e:000043 CFUNC  :fork
c:0003 p:0675 s:0040 e:000039 METHOD /usr/lib/ruby/ignition/cmdgazebo4.rb:456
c:0002 p:0628 s:0019 E:000e48 EVAL   /usr/bin/ign:267 [FINISH]
c:0001 p:0000 s:0003 E:0025c0 (none) [FINISH]

-- Ruby level backtrace information ----------------------------------------
/usr/bin/ign:267:in `<main>'
/usr/lib/ruby/ignition/cmdgazebo4.rb:456:in `execute'
/usr/lib/ruby/ignition/cmdgazebo4.rb:456:in `fork'
/usr/lib/ruby/ignition/cmdgazebo4.rb:460:in `block in execute'
/usr/lib/ruby/ignition/cmdgazebo4.rb:431:in `runServer'
/usr/lib/ruby/ignition/cmdgazebo4.rb:431:in `call'

-- Machine register context ------------------------------------------------
 RIP: 0x00007f8164aab800 RBP: 0x00007f811eaf3650 RSP: 0x00007f811eaf33d8
 RAX: 0x00007f81192070f0 RBX: 0x0000000000000024 RCX: 0x00007f81180008d0
 RDX: 0x0000000000000024 RDI: 0x00007f8119207114 RSI: 0x0000000000000124
  R8: 0x00007f8118f2e532  R9: 0x00007f8164aab800 R10: 0x0000000000000068
 R11: 0x00007f8164af9e60 R12: 0x0000000000000100 R13: 0x00007f811eaf3610
 R14: 0x00007f811eaf3750 R15: 0x00007f8110001238 EFL: 0x0000000000010207

-- C level backtrace information -------------------------------------------
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f8164eeba15) [0x7f8164eeba15]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f8164eebc4c) [0x7f8164eebc4c]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f8164db5874) [0x7f8164db5874]
/usr/lib/x86_64-linux-gnu/libruby-2.5.so.2.5(0x7f8164e7ba32) [0x7f8164e7ba32]
/lib/x86_64-linux-gnu/libc.so.6(0x7f816497c040) [0x7f816497c040]
/lib/x86_64-linux-gnu/libc.so.6(0x7f8164aab800) [0x7f8164aab800]
/usr/lib/x86_64-linux-gnu/libignition-common3-graphics.so.3(0x7f813c1f8c0c) [0x7f813c1f8c0c]
/usr/lib/x86_64-linux-gnu/libignition-common3-graphics.so.3(_ZNK8ignition6common17SkeletonAnimation4NameB5cxx11Ev+0x1d) [0x7f813c1f8c4d]
/usr/lib/x86_64-linux-gnu/libignition-gazebo4-rendering.so.4(_ZN8ignition6gazebo2v412SceneManager11CreateActorEmRKN3sdf3v105ActorEm+0x963) [0x7f811f7243a3]
/usr/lib/x86_64-linux-gnu/libignition-gazebo4-rendering.so.4(_ZN8ignition6gazebo2v410RenderUtil6UpdateEv+0xea7) [0x7f811f6da197]
/usr/lib/x86_64-linux-gnu/ign-gazebo-4/plugins/libignition-gazebo-sensors-system.so(_ZN8ignition6gazebo2v47systems14SensorsPrivate7RunOnceEv+0xa0) [0x7f811f991770]
/usr/lib/x86_64-linux-gnu/ign-gazebo-4/plugins/libignition-gazebo-sensors-system.so(_ZN8ignition6gazebo2v47systems14SensorsPrivate12RenderThreadEv+0xc0) [0x7f811f991ee0]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(0x7f816100e6df) [0x7f816100e6df]
/lib/x86_64-linux-gnu/libpthread.so.0(start_thread+0xdb) [0x7f81647256db]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f8164a5e71f]

-- Other runtime information -----------------------------------------------

* Loaded script: /usr/bin/ign

* Loaded features:

    0 enumerator.so
    1 thread.rb
    2 rational.so
    3 complex.so
[... 1087 lines cut ...]
7f8165227000-7f8165267000 rw-s 00000000 00:05 587                        /dev/nvidiactl
7f8165267000-7f8165287000 rw-s 00000000 00:05 587                        /dev/nvidiactl
7f8165287000-7f81652c7000 rw-s 00000000 00:05 587                        /dev/nvidiactl
7f81652c7000-7f81652e7000 rw-s 00000000 00:05 587                        /dev/nvidiactl
7f81652e7000-7f81653ef000 rw-p 00000000 00:00 0 
7f81653ef000-7f81653f1000 rw-s 00000000 00:05 587                        /dev/nvidiactl
7f81653f1000-7f81653f2000 rw-s 00000000 00:05 587                        /dev/nvidiactl
7f81653f2000-7f81653f3000 rw-s 00000000 00:05 587                        /dev/nvidiactl
7f81653f3000-7f8165403000 rw-s 00000000 00:05 588                        /dev/nvidia0
7f8165403000-7f8165404000 ---p 00000000 00:00 0 
7f8165404000-7f8165408000 rw-p 00000000 00:00 0 
7f8165408000-7f8165409000 r--p 00029000 103:03 107610759                 /lib/x86_64-linux-gnu/ld-2.27.so
7f8165409000-7f816540a000 rw-p 0002a000 103:03 107610759                 /lib/x86_64-linux-gnu/ld-2.27.so
7f816540a000-7f816540b000 rw-p 00000000 00:00 0 
7fff7f905000-7fff80104000 rw-p 00000000 00:00 0                          [stack]
7fff8014d000-7fff80151000 r--p 00000000 00:00 0                          [vvar]
7fff80151000-7fff80153000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 --xp 00000000 00:00 0                  [vsyscall]

[NOTE]
You may have encountered a bug in the Ruby interpreter or extension libraries.
Bug reports are welcome.
For details: http://www.ruby-lang.org/bugreport.html

```

adlarkin, 2021-02-24:
Something else I noticed was that if I create a world with actors via [actors_population.sdf.erb](https://github.com/ignitionrobotics/ign-gazebo/blob/555a84ea54a6f92b46929eb2f1a43daa13b1fadb/examples/worlds/actors_population.sdf.erb), I am able to run a world with actors in it, but I see some other issues.

For example, run the following to create a world with one actor in it:
1. `wget https://github.com/ignitionrobotics/ign-gazebo/blob/555a84ea54a6f92b46929eb2f1a43daa13b1fadb/examples/worlds/actors_population.sdf.erb`
2. `erb r=1 c=1 actors_population.sdf.erb > actors_population.sdf`
3. `ign gazebo -r actors_population.sdf`

A world with a single actor is started, but I see the following error message:
```
[GUI] [Err] [RenderUtil.cc:867] invalid animation update data
```

tadteo, 2021-03-03:
Same in Ubuntu 20.04 LTS

adlarkin, 2021-04-12:
Actors crashing should now be resolved in Edifice if the following PRs are used (releases haven't been made with these changes yet, so users will need to build from source for now):
* https://github.com/ignitionrobotics/ign-common/pull/188
* https://github.com/ignitionrobotics/ign-gazebo/pull/754

Next, these fixes need to be backported to Citadel and Dome. The backporting work has been started here:
* https://github.com/ignitionrobotics/ign-common/pull/199

chapulina, 2021-04-13:
> releases haven't been made with these changes yet

The `ign-common` fix got in before code freeze :wink:

adlarkin, 2021-04-28:
This should now be fixed in Citadel, Dome and Edifice since https://github.com/ignitionrobotics/ign-common/pull/199 was just merged.

closed 2021-04-28 by adlarkin