gazebosim/gz-sim #676: How to implement these parameters in lidar plugin?

addy1997, 2021-03-09:
I am modifying the lidar.hh file for my application. I want to add the following parameters 
```
min_distance
max_distance
low_signal_strength
high_signal_strength

```
Should I create a separate class and add these parameters to a function?

chapulina, 2021-03-09:
> I am modifying the lidar.hh file

Do you mean [this one](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo4/include/ignition/gazebo/components/Lidar.hh)?

> for my application

Do you mean you want this to be available only to your application, or do you want to eventually push it back to Ignition Gazebo?

 ---

Depending on what you need, you may need to make changes in `ign-rendering`, `ign-sensors` and `sdformat` too.

chapulina, 2021-03-09:
Ok, there are many ways you can go about this. You'll probably need to create a new lidar sensor in `ign-sensors`.

Since this is not a feature request or a bug report, I'll close this ticket, and we can continue the conversation on https://answers.gazebosim.org/. Also, please use the ticket templates when ticketing issues in the future.

closed 2021-03-09 by chapulina