gazebosim/gz-sim #679: distortion and irregularities in simulation in ignition gazebo 4

addy1997, 2021-03-11:
## Environment
* OS Version: Ubuntu 18.04 (Virtual Machine)
* Source or binary build?
 binary installed (version below)
```
adwaitnaik@ubuntu:~$ ign gazebo -v
[Msg] Ignition Gazebo GUI    v4.6.0
[Msg] Ignition Gazebo Server v4.6.0

```
## Description
* Expected behavior: The animal should simulate normally as it does in the Blender (version 2.8). Please see the images below

[image]

[image]

* Actual behavior: File format 

* terminal throws the following error 

`[GUI] [Err] [RenderUtil.cc:879] invalid animation update data`

* The animal is distorted and disoriented in the ign gazebo (please see the image below)

[image]

[image]

## Steps to reproduce
Link to the files below (files are in .dae and .sdf format)

* [model file](https://drive.google.com/file/d/1XlaegGLUggGMK_G6P6IJIUK0aAyyb1QM/view?usp=sharing)
* [collada file](https://drive.google.com/file/d/1mTxGHydZLFmnJnfgccide-QnJjJJvYV2/view?usp=sharing)

Thanks

chapulina, 2021-03-11:
Cool model :deer: 

We have some actors that were created on Blender and animate correctly on Ignition, like [Male visitor](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Male%20visitor). The tricky part is that it usually takes us some back-and-forth with various Blender export settings until we get the models to work correctly on Ignition. Once we figure out the fail-proof export settings, we'll write a tutorial for the community.

Meanwhile, I think you could try changing some settings when you export from Blender. You could also try exporting the animation to BVH to see if that helps. It's also possible that our [ColladaLoader](https://github.com/ignitionrobotics/ign-common/blob/ign-common3/graphics/src/ColladaLoader.cc) may need some tweaks to properly support that model.

Please let us know if you get it to work!

addy1997, 2021-03-26:
https://answers.gazebosim.org/question/26731/irregularities-in-simulation-in-ignition-gazebo-4/

closed 2021-12-24 by addy1997

ebasatemesgen, 2025-04-03:
Hi, would you be able to share your 3D model. It will be great to test that out in the new gazebo