gazebosim/gz-sim #687: Failed to update IMU: 16. Entity not found

cocodmdr, 2021-03-16:
Hello,

I am using ROS Foxy with Ignition Dome.

I have this error when I imported the IMU plugin and I cannot find a solution :
> [ign gazebo-1] [Err] [Imu.cc:224] Failed to update IMU: 16. Entity not found.

I am using a fork of Chapulina dolly: [here](https://github.com/cocodmdr/dolly)

When I run without a launch file I don't have this error : 
> ign gazebo -v 4 ~/ws/src/dolly/dolly_ignition/models/tutorial_robot.sdf
And I can see that the IMU is working with this command: 
> ign topic -e -t /imu/data

But when I run within a launch file I have an error:
> ros2 launch dolly_ignition tuto.launch.py

Thank you for your help !

matosinho, 2021-10-01:
I'm having the same problem, but I'm not using ROS. The IMU error appears when I launch an .ign with a robot that has an IMU.

ahcorde, 2021-10-01:
Did you try this fix https://github.com/ignitionrobotics/ign-gazebo/pull/1043 ?

matosinho, 2021-10-03:
I could only test that branch rebased on the main. Even though the rebase had no conflits to be solved, the fix did not work here.