gazebosim/gz-sim #709: Latest Edifice nightlies on Bionic crash at runtime

chapulina, 2021-03-26:
## Environment
* OS Version:  Ubuntu Bionic
* Source or binary build?

<details>
  <summary>Latest debs</summary>
  
```
$ apt list | grep ignition.*installed

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ignition-edifice/unknown,now 0.999.999+nightly+git20210325+1r5aad73ebc1f9c522f89ac4f159a4285e604b3465-1~bionic amd64 [installed]
ignition-tools/unknown,now 1.1.0~pre1-1~bionic amd64 [installed,automatic]
libignition-cmake2-dev/unknown,now 2.7.0~pre2-1~bionic amd64 [installed,automatic]
libignition-common4/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-av/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-av-dev/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-core-dev/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-dev/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-events/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-events-dev/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-graphics/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-graphics-dev/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-profiler/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-common4-profiler-dev/unknown,now 4.0.0~pre1+nightly+git20210325+1rb68fd3a1a037b72524bd5c5557697d0b75132302-1~bionic amd64 [installed,automatic]
libignition-fuel-tools6/unknown,now 6.0.0~pre1+nightly+git20210325+1r2e5a319a1105dcb089660cb95c1fd97de391a99f-1~bionic amd64 [installed,automatic]
libignition-fuel-tools6-dev/unknown,now 6.0.0~pre1+nightly+git20210325+1r2e5a319a1105dcb089660cb95c1fd97de391a99f-1~bionic amd64 [installed,automatic]
libignition-gazebo5/unknown,now 4.999.999+nightly+git20210326+2r2bf29c214bc81d5b75a087f995582d7a82908445-2~bionic amd64 [installed,automatic]
libignition-gazebo5-dev/unknown,now 4.999.999+nightly+git20210326+2r2bf29c214bc81d5b75a087f995582d7a82908445-2~bionic amd64 [installed,automatic]
libignition-gazebo5-plugins/unknown,now 4.999.999+nightly+git20210326+2r2bf29c214bc81d5b75a087f995582d7a82908445-2~bionic amd64 [installed,automatic]
libignition-gui5/unknown,now 5.0.0~pre1-2~bionic amd64 [installed,automatic]
libignition-gui5-dev/unknown,now 5.0.0~pre1-2~bionic amd64 [installed,automatic]
libignition-launch4/unknown,now 3.999.999+nightly+git20210325+1r8bdf0f63cd14380d282d834ed5c03b76c804012b-1~bionic amd64 [installed,automatic]
libignition-launch4-dev/unknown,now 3.999.999+nightly+git20210325+1r8bdf0f63cd14380d282d834ed5c03b76c804012b-1~bionic amd64 [installed,automatic]
libignition-math6/unknown,now 6.8.0~pre3-1~bionic amd64 [installed,automatic]
libignition-math6-dev/unknown,now 6.8.0~pre3-1~bionic amd64 [installed,automatic]
libignition-math6-eigen3-dev/unknown,now 6.8.0~pre3-1~bionic amd64 [installed,automatic]
libignition-msgs7/unknown,now 7.0.0~pre1+nightly+git20210325+1r584fdd89962b336fede8b6703ace159b6eec0737-1~bionic amd64 [installed,automatic]
libignition-msgs7-dev/unknown,now 7.0.0~pre1+nightly+git20210325+1r584fdd89962b336fede8b6703ace159b6eec0737-1~bionic amd64 [installed,automatic]
libignition-physics4/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-core-dev/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-dartsim/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-dartsim-dev/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-dev/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-mesh-dev/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-sdf-dev/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-tpe/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-tpe-dev/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-tpelib/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-physics4-tpelib-dev/unknown,now 4.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-plugin/unknown,now 1.2.0~pre1-1~bionic amd64 [installed,automatic]
libignition-plugin-dev/unknown,now 1.2.0~pre1-1~bionic amd64 [installed,automatic]
libignition-rendering5/unknown,now 5.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-rendering5-core-dev/unknown,now 5.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-rendering5-dev/unknown,now 5.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-rendering5-ogre1/unknown,now 5.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-rendering5-ogre1-dev/unknown,now 5.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-rendering5-ogre2/unknown,now 5.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-rendering5-ogre2-dev/unknown,now 5.0.0~pre1-1~bionic amd64 [installed,automatic]
libignition-sensors5/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-air-pressure/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-air-pressure-dev/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-altimeter/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-altimeter-dev/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-camera/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-camera-dev/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-core-dev/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-depth-camera/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
libignition-sensors5-depth-camera-dev/unknown,now 5.0.0~pre2-1~bionic amd64 [installed,automatic]
[... 63 lines cut ...]
* `ign gazebo 4.999.999+nightly+git20210326+2r2bf29`

This didn't fix it.

I compared the versions on Bionic and Focal, and they're all the same except for `ign-utils`:

Focal:

```
libignition-utils1-dev/unknown,now 1.0.0~pre2+nightly+git20210325+1r5ff07a40ffe111428958e0c79c7ba8b38c090ba5-1~focal amd64 [installed,automatic]
libignition-utils1/unknown,now 1.0.0~pre2+nightly+git20210325+1r5ff07a40ffe111428958e0c79c7ba8b38c090ba5-1~focal amd64 [installed,automatic]
```

Bionic is still stuck in February:

```
libignition-utils1-dev/unknown,now 1.0.0+nightly+git20210225+1rc195c59ce4e03a2ac06aabd0dbeb5dff76142d68-1~bionic amd64 [installed,automatic]
libignition-utils1/unknown,now 1.0.0+nightly+git20210225+1rc195c59ce4e03a2ac06aabd0dbeb5dff76142d68-1~bionic amd64 [installed,automatic]
```

It looks like our uploader is [not uploading the pre-releases](https://build.osrfoundation.org/job/repository_uploader_packages/10912/console), I think it may be because the nightlies were released without the `pre` suffix.

I don't know how the `ign-utils` package could be causing this crash though.

@j-rivero, any ideas?

j-rivero, 2021-03-26:
I don't know what lead to the situation of having a 1.0.0+nightly into the repository but at that point it is completely fine that 1.0.0\~preX are not being uploaded since 1.0.0 stable is greater than 1.0.0\~preX. Mixing nightlies and prereleases could be dangerous since they will need to match the ABIs using during the building process. We can review the situation after the release, be careful when mixing prereleases and nightlies.

chapulina, 2021-03-26:
There was some ABI mismatching between packages. The latest round of nightlies is working on Bionic. We'll trigger new pre-releases today. 

> We can review the situation after the release, be careful when mixing prereleases and nightlies.

+1, let's do that. For now, things are working, so I'll close this.

closed 2021-03-26 by chapulina

chapulina, 2021-03-30:
It looks like we had fixed nightlies, but not pre-releases. The latest `ign-gui 5.0.0~pre2` pre-release does the trick. Just `sudo apt update && sudo apt upgrade`.