gazebosim/gz-sim #712: Capsule geometry is not selectable in 3D view

AndrejOrsula, 2021-03-26:
## Environment
* **OS Version:** Ubuntu 20.04
* **Ignition:** Edifice, built from source
* **ign-gazebo branch:** `main` (2bf29c214bc81d5b75a087f995582d7a82908445), built from source - together with Edifice

## Description
* Expected behavior: Capsule geometry can be selected in 3D view, similar to other geometry types.
* Actual behavior: Capsure cannot be selected in 3D view. It can only be selected via Entity Tree.

## Steps to reproduce
1. `ign gazebo shapes.sdf`
2. Try to select any of the primitives - works except for capsule
3. Try adding new capsule - still does not work

## Output

https://user-images.githubusercontent.com/22929099/112662918-87b53c80-8e58-11eb-9ef5-47747b2ae7e9.mp4

AndrejOrsula, 2021-03-26:
Oh, just realised that this is a duplicate of https://github.com/ignitionrobotics/ign-rendering/issues/283. Closing this issue.

Note to myself: Better check [Core development](https://github.com/orgs/ignitionrobotics/projects/3) instead of repository issues next time.

closed 2021-03-26 by AndrejOrsula

chapulina, 2021-03-26:
Wow the font on your UI is really tiny. Are you on a High DPI display? I thought we had fixed this on https://github.com/ignitionrobotics/ign-gazebo/pull/291

AndrejOrsula, 2021-03-26:
It does not seem to play nicely when I use it inside Docker container (custom image, not using [ign-rocker](https://github.com/adlarkin/ign-rocker)/[gzdev](https://github.com/ignition-tooling/gzdev) - so it is most definitely my fault of some sort). I can read most of the text the very few times I actually use Entity Tree/Component Inspector, so I did not bother investigating it further :smile: - might do later.

When used locally, I have not experienced any issues whatsoever and my eyes are so happy :)

chapulina, 2021-03-27:
Interesting, thanks for the data point.