gazebosim/gz-sim #713: Implementing Ardupilot IR Lock plugin for ignition gazebo

addy1997, 2021-03-26:
## Desired behavior
An IR-based camera for fiducial detection.

## Alternatives considered
No camera /sensor in the present version of ign gazebo satisfies our application

## Implementation suggestion
Here's the header file that I borrowed from the gazebo and changed the interface. The code can be found below

```
#ifndef _IGNITION_GAZEBO_ARDUCOPTERIRLOCK_PLUGIN_HH_
#define _IGNITION_GAZEBO_ARDUCOPTERIRLOCK_PLUGIN_HH_

#include <string>
#include <memory>

#include "igniton/common/Plugin.hh"
#include <ignition/gazebo/System.hh>

namespace ignition 

{
  namespace gazebo 

  {

	// Inline bracket to help Doxygen filtering
    inline namespace IGNITION_GAZEBO_VERSION_NAMESPACE {

    // Forward declare private class
    class IGNITION_GAZEBO_HIDDEN ArduCopterIRLockPluginPrivate;
    

    /// \brief A camera sensor plugin for fiducial detection
    class IGNITION_GAZEBO_VISIBLE ArduCopterIRLockPlugin 

    {
    	/// \brief Constructor
    	public: ArduCopterIRLockPlugin();

    	/// \brief destructor
    	public: virtual  ~ArduCopterIRLockPlugin();

        /// \brief Configure the system
        /// \param[in] _entity The entity this plugin is attached to.
        /// \param[in] _sdf The SDF Element associated with this system plugin.
        /// \param[in] _ecm The EntityComponentManager of the given simulation
        /// instance.
        /// \param[in] _eventMgr The EventManager of the given simulation
        /// instance.
        public: virtual void Configure(
                    const Entity &_entity,
                    const std::shared_ptr<const sdf::Element> &_sdf,
                    EntityComponentManager &_ecm,
                    EventManager &_eventMgr) = 0;

        /// \brief Callback when a new camera frame is available
	    /// \param[in] _image image data
	    /// \param[in] _width image width
	    /// \param[in] _height image height
	    /// \param[in] _depth image depth
	    /// \param[in] _format image format

        public: virtual void OnNewFrame(const unsigned char *_image,
        unsigned int _width, unsigned int _height, unsigned int _depth,
        const std::string &_format);

        /// \brief Publish the result
	    /// \param[in] _name Name of fiducial
	    /// \param[in] _x x position in image
	    /// \param[in] _y y position in image
	    public: virtual void Publish(const std::string &_fiducial, unsigned int _x,
	        unsigned int _y);

	    /// \internal
	    /// \brief Pointer to private data.
	    private: std::unique_ptr<ArduCopterIRLockPluginPrivate> dataPtr;

    };

    }
  }
}
#endif #_IGNITION_GAZEBO_ARDUCOPTERIRLOCK_PLUGIN_HH_
```

## Additional context
I am kinda doubtful about the syntax, this line in particular.    

 ```
public: virtual void Configure(
                    const Entity &_entity,
                    const std::shared_ptr<const sdf::Element> &_sdf,
                    EntityComponentManager &_ecm,
                    EventManager &_eventMgr) = 0;
```

mjcarroll, 2021-03-29:
> An IR-based camera for fiducial detection. Precisely, this paper is what we are trying to implement for our application.

The link to the paper doesn't seem to be working here.

addy1997, 2021-03-29:
@mjcarroll [link](https://drive.google.com/file/d/1N7P9oo7iYgTUpFNKyXNhEFMZ1bmH7XPi/view?usp=sharing)

closed 2021-04-05 by addy1997