gazebosim/gz-sim #726: No output from ign topic when running gpu_lidar examples

ammaar8, 2021-03-30:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?
  binary version 5.0.0~pre1
  

## Description
* Expected behavior: In [gpu_lidar_retro_values_sensor.sdf](https://github.com/ignitionrobotics/ign-gazebo/blob/4ea19561061fece12d09027c226d11cb892ace49/examples/worlds/gpu_lidar_retro_values_sensor.sdf) demo file, on running the example running `ign topic -e -t /lidar` in a different terminal should print lidar messages. 
* Actual behavior: Running `ign topic -e -t /lidar` returns nothing. Even `ign topic -l` returns nothing. I can get the lidar visualization using the visualize lidar plugin using the `lidar` topic. 

## Steps to reproduce

1. Run `ign gazebo gpu_lidar_retro_values_sensor.sdf` in a terminal and play the simulation
2. In a different terminal run `ign topic -e -t /lidar`
3. In another terminal run `ign topic -l`

## Output

chapulina, 2021-06-17:
I can't reproduce this issue with the latest Ignition Edifice debs. Is this still an issue for you, @ammaar8 ?

ammaar8, 2021-06-19:
I tried running it again but still have the same issue. It could be something related to my system. I'm switching to focal so won't be able to test it going forward.

chapulina, 2021-06-21:
Oh ok, so this happens only on Bionic?

I'd be interested to learn if you had more than one Ignition version installed, which can be checked with `ign topic --versions`.

ammaar8, 2021-06-22:
Just tested on Focal and it works fine. 

I checked and turns out I do have two versions installed `4.0.0` and `10.0.0`. Ran with `--force-version 10.0.0` at it works! Sorry for all the trouble.

closed 2021-06-22 by ammaar8

chapulina, 2021-06-22:
Glad to hear it works!