gazebosim/gz-sim #732: Error: ‘std::chrono::steady_clock’ has not been declared

addy1997, 2021-03-31:
I am trying to migrate a plugin for the **actor** to **follow a desired path/trajectory** from (implemented in gazebo) to ignition gazebo. 

While running the header[ file](https://drive.google.com/file/d/1dyGmJevGLsNa0S0ce811E9jYqgSauFft/view?usp=sharing), I got this error

`Error: ‘std::chrono::steady_clock’ has not been declared`

## Environment
* OS Version: Ubuntu 18.04
* Source or binary build?: Ignition 4 (binaries installed)

## Description
* Expected behavior: I am expecting the actor to follow a circular trajectory.

* Actual behavior: In actuality, the ign gazebo opens and closes abruptly. Also, I get the error (mentioned above) on the console. 

# Related files
* [header](https://drive.google.com/file/d/1dyGmJevGLsNa0S0ce811E9jYqgSauFft/view?usp=sharing)
* [main](https://drive.google.com/file/d/1u-zj5duScxrRGtF8i3ynD2yF8C8J-L9e/view?usp=sharing)
* [CmakeList.txt](https://drive.google.com/file/d/1G30clXkMSZZZ_BZmyyFhOLNOgXES1wdZ/view?usp=sharing)

# Additional Context
* I have already raised a question on the forum: [link](https://answers.gazebosim.org/question/26744/plugin-for-actor-to-follow-particular-trajectory/)
* I referred to the ArduPlugin migration doc for relevant instructions to change the header files and syntax. 

Thanks

mjcarroll, 2021-03-31:
In order for the plugin to load, it needs to first be successfully built.  The error you describe sounds like a compilation error, which would prevent the plugin `.so` from being correctly generated.

Can you post more context in the compilation failure?

addy1997, 2021-03-31:
> In order for the plugin to load, it needs to first be successfully built. The error you describe sounds like a compilation error, which would prevent the plugin `.so` from being correctly generated.
> 
> Can you post more context in the compilation failure?

Exactly @mjcarroll , I am not able to build it, let alone test it. Presently, this (the error) is all what I have.

chapulina, 2021-03-31:
@addy1997 , this doesn't sound like a bug on Ignition, but a problem with your plugin. For such questions, we ask that users go to https://answers.gazebosim.org/. This issue tracker is for bug reports and feature requests.

I'm closing this ticket, see you on Gazebo Answers!

closed 2021-03-31 by chapulina