gazebosim/gz-sim #741: Regression in rendering of Collada model

luca-della-vedova, 2021-04-06:
## Environment
Ubuntu 20.04
Edifice, both source and binary

## Description

Regression in rendering of the standing person Collada model from [Fuel](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Standing%20person). The skin appears as a single yellow blob under Edifice but works well in Dome.

## Steps to reproduce

1. Run `ign gazebo` in either dome or edifice
2. Drag and drop the model from Fuel

## Output
In Dome:
[image]
In Edifice:
[image]

iche033, 2021-04-06:
Looks like a 3d emjoi now: :standing_man: 

hmm I can't seem to reproduce this on edifice. I am on source build on 18.04. Tested by dragging the `Standing person` from fuel into ign-gazebo and also tried spawning an sdf file with the model in it. I see the mesh rendered in the image.

luca-della-vedova, 2021-04-07:
Hum I updated my source build and I can't seem to reproduce, must have been because it was outdated :man_shrugging: 
Closing this, apologies for the noise!

closed 2021-04-07 by luca-della-vedova