gazebosim/gz-sim #743: Performance drops significantly when echoing the /world/shapes/dynamic_pose/info topic

adlarkin, 2021-04-06:
## Desired behavior
@iche033 discovered that echoing the `/world/shapes/dynamic_pose/info` topic for a world with a lot of shapes results in a significant RTF decrease. On my machine, I found that my RTF dropped from 90% to 10% when I started to echo this topic (using TPE, and running headless).

## Alternatives considered

TBD

## Implementation suggestion

It seems like [this function](https://github.com/ignitionrobotics/ign-gazebo/blob/ignition-gazebo5_5.0.0/src/systems/scene_broadcaster/SceneBroadcaster.cc#L335) is being called, which causes the slow down. So, I think that we should try to optimize this method.

## Additional context

At first glance, it appears as if this could be related to #711, since the method linked above uses a lot of `Each` calls.

adlarkin, 2021-09-02:
@iche033 I think that we can close this now that #856 is merged. If you look at the new performance testing numbers in the PR description of #856, the `/world/shapes/dynamic_pose/info` RTF is a lot better now. If you think that the new performance of this topic with #856 is good enough, let me know, and I will close this.

iche033, 2021-09-02:
sounds good, closing

closed 2021-09-02 by iche033