gazebosim/gz-sim #744: Improve GUI performance

adlarkin, 2021-04-06:
## Desired behavior

#736 results in good RTF numbers (90-100% on my machine) using TPE in a world of 3000 static shapes when running headless. However, running regular simulation (with the GUI) with the changes in #736 drops RTF to about 30%. There shouldn't be this big of a disparity between headless and GUI simulation. I'd like to figure out why using the GUI is causing such a noticeable performance drop.

## Alternatives considered
N/A

## Implementation suggestion
TBD - I will update this once I can figure out some reasons for the slowdown.

## Additional context
None at the moment - I will update this if I find anything else related to this issue that is worth mentioning (could be related to #711).

chapulina, 2021-04-06:
> It looks like the scene broadcaster system doesn't do much if there is no GUI:

Yeah I believe the `SceneBroadcaster` is the main reason for the slowdown when running with the GUI. See https://github.com/ignitionrobotics/ign-gazebo/issues/556 for an idea that can shortcut the use of the `SceneBroadcaster`, by running server and client on the same process instead.

chapulina, 2021-05-06:
Let's follow up in #556

closed 2021-05-06 by chapulina