gazebosim/gz-sim #757: API for LINK property "AddForceAtRelativePosition"

addy1997, 2021-04-11:
This ticket is raised with reference to this #375 tutorial. I couldn't find a suitable replacement for the "AddForceAtRelativePosition" property of Link API. 

## Desired behavior

I am writing a wind plugin for pheonixDrone for calculating wind force. 
In classical Gazebo, it can be written like this 

```
// Calculate the wind force.
  double wind_strength = wind_force_mean_;

  math::Vector3 wind = wind_strength * wind_direction_;

  // Apply a force from the constant wind to the link.

  link_->AddForceAtRelativePosition(wind, xyz_offset_);
```

## Alternatives considered
Nope, couldn't find suitable alternatives for this.

diegoferigo, 2021-04-11:
This is not currently possible with the existing [Link APIs](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/include/ignition/gazebo/Link.hh). You need to use either `Link::AddWorldForce` or `Link::AddWorldWrench`. Be very careful when using any of these methods because they need the force / wrench expressed in different frames (xref https://github.com/ignitionrobotics/ign-gazebo/issues/201).

In order to express a force in different frames, you can check the following logic:

https://github.com/ignitionrobotics/ign-gazebo/blob/26832764749291b6ea22b525262136ecbb63fbb9/src/Link.cc#L291-L297

Another example is [this downstream project](https://github.com/robotology/gym-ignition/blob/3b1787912ed8a86d93ce1cfbe9d72c53380468f6/cpp/scenario/gazebo/src/Link.cpp#L529-L557).

addy1997, 2021-04-11:
@diegoferigo , thanks.

closed 2021-04-11 by addy1997