gazebosim/gz-sim #759: Symbols of detail::View class are public

adlarkin, 2021-04-13:
When working on #752, it became apparent that [the `detail::View` class has public visibility](https://github.com/ignitionrobotics/ign-gazebo/blob/ignition-gazebo5_5.0.0/include/ignition/gazebo/detail/View.hh#L48). This makes it difficult to change things in this class since it will break ABI. Considering the fact that `detail` classes were never intended to be used by external users, we should probably make the visibility of this class hidden. I believe that this could be as simple as switching `detail::View`'s visibility to `IGNITION_GAZEBO_HIDDEN` for Fortress, but as @iche033 pointed out, we will need to test this to ensure that other platforms and libraries aren't broken by this change.

@chapulina, what do you think about this? Are there any reasons why we should keep the visibility of `detail::View` (and anything else in `detail`) public, or are we okay to make the change to hidden?

chapulina, 2021-04-13:
Ahh that's unfortunate. We should hide everything inside `detail`.

adlarkin, 2021-04-30:
I'm closing this since #788 was merged (it should have closed automatically since this issue was linked to #788, but for some reason, it didn't :man_shrugging:)

PR #788 (2021-04-29): explain why detail::View symbols are visible

# 🦟 Bug fix

Fixes #759

## Summary
~~This hides the symbols of the `detail::View` class so that we can make changes to it as needed without breaking ABI. Since this is an ABI-breaking change, this has to be done in fortress.~~

**Edit:** the symbol visibility for `detail::View` cannot be hidden. See the comments below for an explanation.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobot [... cut]
closed 2021-04-30 by adlarkin