gazebosim/gz-sim #769: error: ' _ecm.Component<components::JointPosition>' method

addy1997, 2021-04-15:
> OS: Ubuntu 18.04 (installed using Virtual Machine)
> binary/source: source 
> version

```
[Msg] Ignition Gazebo Server v4.7.0
[Msg] Ignition Gazebo GUI    v4.7.0
```

I got this error while calculating joints position using the `_ecm.Component<components::JointPosition>` method.

```
error: cannot convert ‘const ignition::gazebo::v3::components::Component<std::vector<double>, ignition::gazebo::v3::components::JointPositionTag, ignition::gazebo::v3::serializers::VectorDoubleSerializer>*’ to ‘float’ in initialization

 float leftAileron = _ecm.Component<components::JointPosition>(this->joint[2500]);
```

**Associated code snippet**

```
void CessnaPlugin::PublishState(const EntityComponentManager &_ecm)

{
    IGN_PROFILE("CessnaPlugin::PublishState");
  
    // All the lines are shown below throw this error

    float propellerRpms = _ecm.ComponentData<components::JointVelocity>(this->joint[6])/(2.0*M_PI)*60.0;
    float propellerSpeed = propellerRpms / this->propellerMaxRpm;
    float leftAileron = _ecm.Component<components::JointPosition>(this->joint[2500]);
    float leftFlap =  _ecm.Component<components::JointPosition>(this->joint[1]);
    float rightAileron =  _ecm.Component<components::JointPosition>(this->joint[2]);
    float rightFlap =  _ecm.Component<components::JointPosition>(this->joint[3]);
    float elevators =  _ecm.Component<components::JointPosition>(this->joint[4]);
    float rudder =  _ecm.Component<components::JointPosition>(this->joint[5]);
}
```

**Additional Context**

I am trying to get the values in **float**. [please refer [cessna.proto](https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs7/proto/ignition/msgs/cessna.proto) for this.]

This method `_ecm.Component<components::JointPosition>` has 'std::vector(double)' type which can't be converted to 'float'.

Note: In the code snippet, the **4** in `this->joint[4] ` comes from here

Cessna parameters as described in plugin class

```
class ignition::gazebo::systems::CessnaPluginPrivate
{
   public: static const unsigned int kLeftAileron  = 0;
       
    public: static const unsigned int kLeftFlap     = 1;
         
    public: static const unsigned int kRightAileron = 2;
        
    public: static const unsigned int kRightFlap    = 3;
        
    public: static const unsigned int kElevators    = 4;
        
    public: static const unsigned int kRudder       = 5;
        
    public: static const unsigned int kPropeller    = 6;
};
```
Since the automatically generated file **cessna.pb.h** can't be edited, I could find any solution for this. Any suggestion is appreciated. 
Thanks

diegoferigo, 2021-04-15:
A joint in its most generic form can have more than one degree of freedom. This is the reason why the `JointPosition` component stores a vector:

https://github.com/ignitionrobotics/ign-gazebo/blob/a41cf80049c07977079d0b215398dc4225b63038/include/ignition/gazebo/components/JointPosition.hh#L38-L39

In 99% of cases, you're using either a revolute or prismatic joint, therefore the vector contains only one element:

```cpp
const auto& jointPosition = _ecm.ComponentData<components::JointVelocity>(jointEntity);
assert(jointPosition.size() == 1);
const float position = jointPosition[0];
```

addy1997, 2021-04-15:
Thanks @diegoferigo

addy1997, 2021-04-21:
Hello @diegoferigo 

I am trying to use `_ecm.setComponentData()` in my gimbal plugin. The error is in this line

    auto error = (position - this->dataPtr->command);

**Complete error**

     error: invalid operands of types ‘const ignition::gazebo::v5::components::Component<std::vector<double>, ignition::gazebo::v5::components::JointPositionTag, ignition::gazebo::v5::serializers::VectorDoubleSerializer>*’ and ‘double’ to binary ‘operator-’

**Code snippet**

    void GimbalSmall2dPlugin::PostUpdate(const UpdateInfo &_info,
        	const EntityComponentManager &_ecm)
    
    {
    	IGN_PROFILE("GimbalSmall2dPlugin::PostUpdate");
    
    	auto position = _ecm.Component<components::JointPosition>(this->dataPtr->tiltJoint[0]);
    
    
    	if (_info.simTime > this->dataPtr->lastUpdateTime)
    	{
    
    		auto dt = (this->dataPtr->lastUpdateTime - _info.simTime);
    		auto error = (position - this->dataPtr->command);
    		auto force = this->dataPtr->pid.Update(error, dt);
    		_ecm.SetComponentData(this->dataPtr->tiltJoint, components::JointForceCmd(force));
    		this->dataPtr->lastUpdateTime = _info.simTime;
    
    	}
    
    	static int i = 1000;
    	if(++i > 100)
    	{
    		i=0;
    		std::stringstream ss;
    		ss << position;
    		msgs::StringMsg m;
    		m.set_data(ss.str());
    		this->dataPtr->pub.Publish(m);
    	}
    
    	IGN_PROFILE_END();
    }

Any suggestion for this?

closed 2021-04-22 by addy1997