gazebosim/gz-sim #771: Ignition Gazebo 3 Segfault in test INTEGRATION_physics_system

sloretz, 2021-04-16:
## Environment
* OS Version:  OSX
* Source or binary build?
   Source
    
<details>
  
```
# paste log here
```

</details>

## Description
https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/131

* Expected behavior:  The test would not segfault
* Actual behavior:  The test segfaulted

## Steps to reproduce

1. Run this CI job https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/131/

## Output

<details><summary>Test output</summary>

```
test 111
        Start 111: INTEGRATION_physics_system

111: Test command: /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/build/bin/INTEGRATION_physics_system "--gtest_output=xml:/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/build/test_results/INTEGRATION_physics_system.xml"
111: Test timeout computed to be: 240
111: Running main() from /Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/ign-gazebo/test/gtest/src/gtest_main.cc
111: [==========] Running 12 tests from 1 test suite.
111: [----------] Global test environment set-up.
111: [----------] 12 tests from PhysicsSystemFixture
111: [ RUN      ] PhysicsSystemFixture.CreatePhysicsWorld
111: [Msg] Loading SDF world file[/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo3-homebrew-amd64/ign-gazebo/test/worlds/shapes.sdf].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [17459188283658606303] / [ign_gazebo_components.RenderEngineServerPlugin].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [8705992680619689917] / [ign_gazebo_components.RenderEngineGuiPlugin].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [17605309075052480649] / [ign_gazebo_components.Atmosphere].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [8064491505919932473] / [ign_gazebo_components.Level].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [2668898242563798256] / [ign_gazebo_components.DefaultLevel].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [11371360182141354106] / [ign_gazebo_components.LevelEntityNames].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [4981278897826323946] / [ign_gazebo_components.WorldSdf].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [6687176221774458630] / [ign_gazebo_components.Model].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [6612894081701502240] / [ign_gazebo_components.Pose].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [8546580419506082455] / [ign_gazebo_components.Static].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [9712747055438129860] / [ign_gazebo_components.WindMode].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [5661073481138181711] / [ign_gazebo_components.SelfCollide].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [11683062252779233161] / [ign_gazebo_components.SourceFilePath].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [5081358965268446661] / [ign_gazebo_components.Link].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [8112400427272910195] / [ign_gazebo_components.Inertial].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [16454635107327670381] / [ign_gazebo_components.Visual].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [13011964647677164955] / [ign_gazebo_components.CastShadows].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [13440282432131634483] / [ign_gazebo_components.Transparency].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [5522706856600206064] / [ign_gazebo_components.LaserRetro].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [17121648710877364109] / [ign_gazebo_components.Geometry].
111: [Dbg] [EntityComponentManager.cc:632] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
[... 139 lines cut ...]
111: [Msg] Serving world names on [/gazebo/worlds]
111: [Msg] Resource path add service on [/gazebo/resource_paths/add].
111: [Msg] Resource path get service on [/gazebo/resource_paths/get].
111: [Msg] Resource paths published on [/gazebo/resource_paths].
111: [Msg] Found no publishers on /stats, adding root stats topic
111: [Msg] Found no publishers on /clock, adding root clock topic
111: [Dbg] [SimulationRunner.cc:422] Creating PostUpdate worker threads: 3
111: [Dbg] [SimulationRunner.cc:435] Creating postupdate worker thread (0)
111: [Dbg] [SimulationRunner.cc:435] Creating postupdate worker thread (1)
111: [Msg] Serving scene information on [/world/default/scene/info]
111: [Msg] Serving graph information on [/world/default/scene/graph]
111: [Msg] Serving full state on [/world/default/state]
111: [Msg] Serving full state (async) on [/world/default/state_async]
111: [Msg] Publishing scene information on [/world/default/scene/info]
111: [Msg] Publishing entity deletions on [/world/default/scene/deletion]
111: [Msg] Publishing state changes on [/world/default/state]
111: [Msg] Publishing pose messages on [/world/default/pose/info]
111: [Msg] Publishing dynamic pose messages on [/world/default/dynamic_pose/info]
111: [Dbg] [SimulationRunner.cc:451] Exiting postupdate worker thread (0)
111: [Dbg] [SimulationRunner.cc:451] Exiting postupdate worker thread (1)
111: [       OK ] PhysicsSystemFixture.CanonicalLink (189 ms)
111: [ RUN      ] PhysicsSystemFixture.NonDefaultCanonicalLink
111/156 Test #111: INTEGRATION_physics_system ...........................***Exception: SegFault  1.99 sec
```
</details>

jacobperron, 2021-07-19:
Another segfault in the same test suite, but a different test: https://build.osrfoundation.org/job/ignition_gazebo-ci-ign-gazebo5-homebrew-amd64/26/

```
133: [ RUN      ] PhysicsSystemFixture.IncludeNestedModelTPE
133: [Msg] Loading SDF world file[/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-homebrew-amd64/ign-gazebo/test/worlds/include_nested_models.sdf].
133: Warning [Utils.cc:121] [/sdf/model:/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-homebrew-amd64/ign-gazebo/test/worlds/models/include_nested/model.sdf]: Found more than one of model for <include>. This is unsupported and in future versions of libsdformat will become an error[Dbg] [EntityComponentManager.cc:688] Using components of type [2251689575469537287] / [ign_gazebo_components.World].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [13994732549916512682] / [ign_gazebo_components.Name].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [12592746352568925681] / [ign_gazebo_components.Gravity].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [2188341333082264598] / [ign_gazebo_components.Physics].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [13224937992534617849] / [ign_gazebo_components.MagneticField].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [3630648173860223239] / [ign_gazebo_components.PhysicsEnginePlugin].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [17459188283658606303] / [ign_gazebo_components.RenderEngineServerPlugin].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [8705992680619689917] / [ign_gazebo_components.RenderEngineGuiPlugin].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [8753193699724811771] / [ign_gazebo_components.Wind].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [12173050716021724529] / [ign_gazebo_components.WorldLinearVelocity].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [15943768124495574352] / [ign_gazebo_components.WorldLinearVelocitySeed].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [3297509811873971798] / [ign_gazebo_components.ParentEntity].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [17100615127981600159] / [ign_gazebo_components.Scene].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [17605309075052480649] / [ign_gazebo_components.Atmosphere].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [8064491505919932473] / [ign_gazebo_components.Level].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [2668898242563798256] / [ign_gazebo_components.DefaultLevel].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [11371360182141354106] / [ign_gazebo_components.LevelEntityNames].
133: [Dbg] [Physics.cc:687] Loaded [ignition::physics::tpeplugin::Plugin] from library [/usr/local/Cellar/ignition-physics4/4.2.0/lib/ign-physics-4/engine-plugins/libignition-physics-tpe-plugin.dylib]
133: [Dbg] [SimulationRunner.cc:851] Loaded system [ignition::gazebo::systems::Physics] for entity [1]
133: [Dbg] [SimulationRunner.cc:851] Loaded system [ignition::gazebo::systems::SceneBroadcaster] for entity [1]
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [4981278897826323946] / [ign_gazebo_components.WorldSdf].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [6687176221774458630] / [ign_gazebo_components.Model].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [6612894081701502240] / [ign_gazebo_components.Pose].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [8546580419506082455] / [ign_gazebo_components.Static].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [9712747055438129860] / [ign_gazebo_components.WindMode].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [5661073481138181711] / [ign_gazebo_components.SelfCollide].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [11683062252779233161] / [ign_gazebo_components.SourceFilePath].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [5081358965268446661] / [ign_gazebo_components.Link].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [8112400427272910195] / [ign_gazebo_components.Inertial].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [16454635107327670381] / [ign_gazebo_components.Visual].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [13011964647677164955] / [ign_gazebo_components.CastShadows].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [13440282432131634483] / [ign_gazebo_components.Transparency].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [5453622280849253520] / [ign_gazebo_components.VisibilityFlags].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [17121648710877364109] / [ign_gazebo_components.Geometry].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [9853217982010720764] / [ign_gazebo_components.Material].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [17938588655714334139] / [ign_gazebo_components.Collision].
133: [Dbg] [EntityComponentManager.cc:688] Using components of type [9225962031573086509] / [ign_gazebo_components.CollisionElement].
[... cut ...]
133: [Msg] Resource path get service on [/gazebo/resource_paths/get].
133: [Msg] Resource paths published on [/gazebo/resource_paths].
133: Warning [Utils.cc:121] [/sdf/model:/Users/jenkins/workspace/ignition_gazebo-ci-ign-gazebo5-homebrew-amd64/ign-gazebo/test/worlds/models/include_nested/model.sdf]: Found more than one of model for <include>. This is unsupported and in future versions of libsdformat will become an error
133/182 Test #133: INTEGRATION_physics_system .............................***Exception: SegFault  5.79 sec
```

azeey, 2023-03-17:
This is likely fixed by #1836. We can re-open if necessary.

closed 2023-03-17 by azeey