gazebosim/gz-sim #773: GUI crashes randomly

doisyg, 2021-04-18:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build: both binary and source for citadel, dome and edifice. 

* Tested on 2 systems with/without a nvidia GPU.
    - Rendering plugin: [ogre2]. 
  
    
<details>

GDB backtrace

```
# [New Thread 0x7fff9cff9700 (LWP 2244306)]
free(): invalid pointer

Thread 13 "ruby" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffcd88d700 (LWP 2242584)]
--Type <RET> for more, q to quit, c to continue without paging--c
tcache_get (tc_idx=<optimized out>) at malloc.c:2937
2937	malloc.c: No such file or directory.
(gdb) backtrace
#0  tcache_get (tc_idx=<optimized out>) at malloc.c:2937
#1  __GI___libc_malloc (bytes=31) at malloc.c:3051
#2  0x00007ffff30ed241 in QArrayData::allocate(unsigned long, unsigned long, unsigned long, QFlags<QArrayData::AllocationOption>) ()
    at /lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007ffff30ef236 in QByteArray::QByteArray(char const*, int) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007fffb14b2a96 in std::pair<int, QByteArray>::pair<int, char const*, true>(std::pair<int, char const*>&&)
    (__p=..., this=0x7fffcd88cc00) at /usr/include/c++/9/bits/move.h:74
#5  ignition::gazebo::ComponentsModel::RoleNames() ()
    at /home/gd/workspace/src/ign-gazebo/src/gui/plugins/component_inspector/ComponentInspector.cc:328
#6  0x00007fffb14b68d4 in ignition::gazebo::ComponentInspector::Update(ignition::gazebo::v5::UpdateInfo const&, ignition::gazebo::v5::EntityComponentManager&) (this=0x5555573d1340, _ecm=...)
    at /home/gd/workspace/src/ign-gazebo/src/gui/plugins/component_inspector/ComponentInspector.cc:460
#7  0x00007ffff3a7a703 in ignition::gazebo::v5::GuiRunner::Implementation::UpdatePlugins() (this=0x5555569dfa40)
    at /home/gd/workspace/src/ign-gazebo/src/gui/GuiRunner.cc:208
#8  0x00007ffff3a7a812 in ignition::gazebo::v5::GuiRunner::<lambda()>::operator() (__closure=0x555556d15918)
    at /usr/include/c++/9/bits/unique_ptr.h:360
#9  std::__invoke_impl<void, ignition::gazebo::v5::GuiRunner::GuiRunner(const string&)::<lambda()> > (__f=...)
    at /usr/include/c++/9/bits/invoke.h:60
#10 std::__invoke<ignition::gazebo::v5::GuiRunner::GuiRunner(const string&)::<lambda()> > (__fn=...)
    at /usr/include/c++/9/bits/invoke.h:95
#11 std::thread::_Invoker<std::tuple<ignition::gazebo::v5::GuiRunner::GuiRunner(const string&)::<lambda()> > >::_M_invoke<0>
    (this=0x555556d15918) at /usr/include/c++/9/thread:244
#12 std::thread::_Invoker<std::tuple<ignition::gazebo::v5::GuiRunner::GuiRunner(const string&)::<lambda()> > >::operator()
    (this=0x555556d15918) at /usr/include/c++/9/thread:251
#13 std::thread::_State_impl<std::thread::_Invoker<std::tuple<ignition::gazebo::v5::GuiRunner::GuiRunner(const string&)::<lambda()> > > >::_M_run(void) (this=0x555556d15910) at /usr/include/c++/9/thread:195
#14 0x00007ffff2f18d84 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#15 0x00007ffff7a24609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#16 0x00007ffff7b60293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

```

</details>

## Description
* Expected behavior: GUI is running without crashing
* Actual behavior: GUI crashes after a random amount of time
## Steps to reproduce

1. (gdb) r /path/to/exec/ign gazebo -g -v 4
2.  (gdb) /path/to/exec/ign -s -v 4 -r shapes.sdf
3. Wait

## Output
See backtrace above in details (`Qt::mightBeRichText(QString const&) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5` ??) and 
[ogre2.log](https://github.com/ignitionrobotics/ign-gazebo/files/6332287/ogre2.log)

MarqRazz, 2021-06-04:
I am having a similar issue where Ignition dies after a few minutes when running `ign gazebo` with nothing added to the world and get this message.
```
SEGV received in SEGV handler
/usr/lib/ruby/ignition/cmdgazebo5.rb:438: [BUG] Segmentation fault at 0x0000000000000000
ruby 2.7.0p0 (2019-12-25 revision 647ee6f091) [x86_64-linux-gnu]

-- Control frame information -----------------------------------------------
```

If I am running my robot in simulation it also dies after a few min but prints out this log.
[seg_fault.log](https://github.com/ignitionrobotics/ign-gazebo/files/6599569/seg_fault.log)

Ubuntu 20.04 with Edifice binary install
Graphics card: GeForce GTX 1060 Mobile running nvidia-driver-460
I have a fellow engineer that is experiencing the same issues but he is running in a Docker container.

If this is not related please let me know and I will create a new issue.

chapulina, 2021-06-15:
@MarqRazz , your backtrace looks similar to the one in #722, pointing to `Qt::mightBeRichText`