gazebosim/gz-sim #785: Adding video overlay feature to the CameraVideoRecorder plugin

addy1997, 2021-04-25:
## Desired behavior
In the [CameraVideoRecorder](https://github.com/ignitionrobotics/ign-gazebo/blob/ign-gazebo5/src/systems/camera_video_recorder/CameraVideoRecorder.hh) plugin, video is recorded using this service https://github.com/ignitionrobotics/ign-gazebo/blob/51befb46a0159c353aec59a2060df55d3f85e48c/src/systems/camera_video_recorder/CameraVideoRecorder.cc#L83 which is quite slow.

Therefore, I wanted to add this feature called [video overlay](https://en.wikipedia.org/wiki/Video_overlay) to the plugin to make the process of video recording faster.

## Alternatives considered
We tried implementing a plugin using OpenCV and CameraSensor to implement the same but the library lacks some features. Displaying the image taken by the camera using OpenCV had some issues like these [wobbling issue](https://drive.google.com/file/d/16N6x3quAylhq7LR9K-u2Gz07UXXX3xjv/view?usp=sharing)

## Implementation suggestion
This is based on my understanding of the software and may not be the exact suggestion. If we can include a `node `that `bypasses the physics engines` and directly `displays the video with another window` (apart from the GUI). 

In short, we can have two windows, the regular ign gazebo one and another window solely for displaying the video. Something like this

[image]

Any suggestion is appreciated. 

Thanks

chapulina, 2021-04-26:
> quite slow for our application

You mean that the process of recording the video is slow, or the recorded video shows a slow simulation?

Are you using the `use_sim_time` and / or `lockstep` options explained on the [Video recorded tutorial](https://ignitionrobotics.org/api/gazebo/5.0/videorecorder.html)?

addy1997, 2021-04-27:
> > quite slow for our application
> 
> You mean that the process of recording the video is slow, or the recorded video shows a slow simulation?
> 
> Are you using the `use_sim_time` and / or `lockstep` options explained on the [Video recorder tutorial](https://ignitionrobotics.org/api/gazebo/5.0/videorecorder.html)?

Yes Ma'am, I have referred to both the tutorials, even this [one](https://ignitionrobotics.org/api/common/3.10/hw-encoding.html). So, I have tried both the options, i.e., recording the video in `use_sim_time ` and `lockstep` on two different machines.

I reckon there is some problem with `identifying the encoder`, not sure though. Well, it shouldn't be coz our setup, RAM, workstaions `(NVIDIA RTX A6000 GPU)` are working fine.

addy1997, 2021-05-04:
The process of recording the video is slow. I am using both  ```use_sim_time``` and ```lockstep```. I referred to [this ](https://ignitionrobotics.org/api/common/3.10/hw-encoding.html) tutorial as well but couldn't get rid of the errors.

closed 2021-05-06 by addy1997