gazebosim/gz-sim #806: Components registered by multiple systems cause segfaults on macOS

azeey, 2021-05-05:
## Environment
* OS Version: macOS Mojave
* Source or binary build? source, ign-gazebo3

## Description
* Expected behavior: No segfault
* Actual behavior:  If a component's header file is included by a system, but not by the translation units that comprise `libignition-gazebo.so`, the component is registered only when that system is loaded. Having two such systems can cause segfaults on macOS. This is manifested in number of flaky tests documented in #444 and other issues. From what I can tell, the problem arises when a test with two or more test cases that create a server and load worlds with those systems, that is

```c++

TEST(Foo, TEST1)
{
  Server server(...); // Load world with two or more systems that include a component's header
  server.Run(...);
  ...
}
 
TEST(Foo, TEST2)
{
  Server server(...);  // Load world with two or more systems that include a component's header
  server.Run(...);
  ...
}
```

An example is test/integration/joint_state_publisher_system.cc

## Steps to reproduce

1. Build ign-gazebo with Address Sanitizer
2. Run `INTEGRATION_joint_state_publisher_system`

## Output

Address sanitizer output from running `INTEGRATION_joint_state_publisher_system`

```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==89646==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x00010338e203 bp 0x7ffeeccaa870 sp 0x7ffeeccaa6c0 T0)
==89646==The signal is caused by a READ memory access.
==89646==Hint: address points to the zero page.
    #0 0x10338e202 in ignition::gazebo::v3::components::Factory::NewStorage(unsigned long long const&) Factory.hh:259
    #1 0x10338b085 in ignition::gazebo::EntityComponentManagerPrivate::CreateComponentStorage(unsigned long long) EntityComponentManager.cc:622
    #2 0x10338a3aa in ignition::gazebo::v3::EntityComponentManager::CreateComponentImplementation(unsigned long long, unsigned long long, ignition::gazebo::v3::components::BaseComponent const*) EntityComponentManager.cc:483
    #3 0x11a3c82e6 in std::__1::pair<unsigned long long, int> ignition::gazebo::v3::EntityComponentManager::CreateComponent<ignition::gazebo::v3::components::Component<std::__1::vector<double, std::__1::allocator<double> >, ignition::gazebo::v3::components::JointPositionTag, ignition::gazebo::v3::serializers::VectorDoubleSerializer> >(unsigned long long, ignition::gazebo::v3::components::Component<std::__1::vector<double, std::__1::allocator<double> >, ignition::gazebo::v3::components::JointPositionTag, ignition::gazebo::v3::serializers::VectorDoubleSerializer> const&) EntityComponentManager.hh:86
    #4 0x11a3c5f68 in ignition::gazebo::v3::systems::JointStatePublisher::CreateComponents(ignition::gazebo::v3::EntityComponentManager&, unsigned long long) JointStatePublisher.cc:110
    #5 0x11a3c4b26 in ignition::gazebo::v3::systems::JointStatePublisher::Configure(unsigned long long const&, std::__1::shared_ptr<sdf::v9::Element const> const&, ignition::gazebo::v3::EntityComponentManager&, ignition::gazebo::v3::EventManager&) JointStatePublisher.cc:71
    #6 0x103acf559 in ignition::gazebo::v3::SimulationRunner::LoadPlugin(unsigned long long, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::shared_ptr<sdf::v9::Element> const&) SimulationRunner.cc:769
    #7 0x103aba5d1 in ignition::gazebo::v3::SimulationRunner::LoadPlugins(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&) SimulationRunner.cc:903
    #8 0x103b61e2e in decltype(*(std::__1::forward<ignition::gazebo::v3::SimulationRunner*&>(fp0)).*fp(std::__1::forward<unsigned long long>(fp1), std::__1::forward<std::__1::shared_ptr<sdf::v9::Element> >(fp1))) std::__1::__invoke<void (ignition::gazebo::v3::SimulationRunner::*&)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), ignition::gazebo::v3::SimulationRunner*&, unsigned long long, std::__1::shared_ptr<sdf::v9::Element>, void>(void (ignition::gazebo::v3::SimulationRunner::*&&&)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), ignition::gazebo::v3::SimulationRunner*&&&, unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&) type_traits:4302
    #9 0x103b61c59 in std::__1::__bind_return<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), std::__1::tuple<ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1>, std::__1::placeholders::__ph<2> >, std::__1::tuple<unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&>, __is_valid_bind_return<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), std::__1::tuple<ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1>, std::__1::placeholders::__ph<2> >, std::__1::tuple<unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&> >::value>::type std::__1::__apply_functor<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), std::__1::tuple<ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1>, std::__1::placeholders::__ph<2> >, 0ul, 1ul, 2ul, std::__1::tuple<unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&> >(void (ignition::gazebo::v3::SimulationRunner::*&)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), std::__1::tuple<ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1>, std::__1::placeholders::__ph<2> >&, std::__1::__tuple_indices<0ul, 1ul, 2ul>, std::__1::tuple<unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&>&&) functional:2644
    #10 0x103b61b64 in std::__1::__bind_return<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), std::__1::tuple<ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1>, std::__1::placeholders::__ph<2> >, std::__1::tuple<unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&>, __is_valid_bind_return<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), std::__1::tuple<ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1>, std::__1::placeholders::__ph<2> >, std::__1::tuple<unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&> >::value>::type std::__1::__bind<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&>::operator()<unsigned long long, std::__1::shared_ptr<sdf::v9::Element> >(unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&) functional:2677
    #11 0x103b619f6 in decltype(std::__1::forward<std::__1::__bind<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&>&>(fp)(std::__1::forward<unsigned long long>(fp0), std::__1::forward<std::__1::shared_ptr<sdf::v9::Element> >(fp0))) std::__1::__invoke<std::__1::__bind<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&>&, unsigned long long, std::__1::shared_ptr<sdf::v9::Element> >(std::__1::__bind<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&>&&&, unsigned long long&&, std::__1::shared_ptr<sdf::v9::Element>&&) type_traits:4361
    #12 0x103b61986 in void std::__1::__invoke_void_return_wrapper<void>::__call<std::__1::__bind<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long long, std::__1::shared_ptr<sdf::v9::Element> const&), ignition::gazebo::v3::SimulationRunner*, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&>&, unsigned long long, std::__1::shared_ptr<sdf::v9::Element> >(std::__1::__bind<void (ignition::gazebo::v3::SimulationRunner::*)(unsigned long l [... cut ...]
rdi = 0x00007ffeeccaa720  rsi = 0x00006020000bbb10  rbp = 0x00007ffeeccaa870  rsp = 0x00007ffeeccaa6c0
 r8 = 0x000000010402ba30   r9 = 0x000000010402ba30  r10 = 0x05c5fa1df8df0041  r11 = 0x00000000000001c0
r12 = 0x0000100000000000  r13 = 0x00001fffdd995568  r14 = 0x00001fffdd995514  r15 = 0x00001fffdd995560
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV Factory.hh:259 in ignition::gazebo::v3::components::Factory::NewStorage(unsigned long long const&)
==89646==ABORTING
```

chapulina, 2021-05-06:
Can you help me understand the result of the address sanitizer?

What is happening when 2 systems are loaded in this case, is the same component registered twice into different static maps?

azeey, 2021-11-03:
Sorry for the long delay in responding. I'm not sure exactly what's happening so I didn't feel encouraged to respond, but I am seeing the same issue manifested in a different way in #869. I think what's happening is that each system gets its own copy of the component and they are treated as though they are different classes in macOS. This means one component will have a different typeinfo data structures in each system and dynamic casting from a base class to a derived class fails where we expect it to succeed.

azeey, 2024-01-31:
Fixed by #1836.

PR #1649 (2022-09-01): Use RTLD_NODELETE=true when loading libraries
Fixes (Potentially): https://github.com/gazebosim/gz-sim/issues/806

## Summary

Gazebo Garden [crashes](https://github.com/gazebosim/gz-sim/issues/1443) at exit because the destructor for an `EventT` object created by a gui plugin is accesses after the plugin is unloaded. For that particular crash, the plugin in question is in gz-gui, so I've submitted a fix there (https://github.com/gazebosim/gz-gui/pull/469). This PR is proactively using `RTLD_NODELETE` to prevent future bugs of this nature. 

I believe this might also fix #806 because if the sh [... cut]
closed 2024-01-31 by azeey